LiberTEM/LiberTEM #400: Proper FFT backend selection

sk1p, 2019-07-20:
Need to have a more central point of selecting the FFT backend, instead of sprinkling the code with `try..import..except` statements. 

See also:

Comment in blobfinder: https://github.com/LiberTEM/LiberTEM/blob/7da5db3048b9bfeb2720c2e113348d5f4ca8a34a/src/libertem/udf/blobfinder.py#L14

HPC FFT lib fluidfft: https://bitbucket.org/fluiddyn/fluidfft/src/default/

sk1p, 2019-11-11:
As an update, the GSOC project did happen: https://blogs.python-gsoc.org/en/peterbell10s-blog/

That work should be released as part of scipy 1.4 AFAIK; when that happens we can fix this issue.

sk1p, 2019-11-12:
We should also think about providing a "registry"-like interface, where people can register any third-party package as fft backend - that way people can also use mkl fft, which has a proprietary license.

uellue, 2022-07-28:
As a side note, pyfftw created issues on the threaded executor -- one might want to consider that as well in the backend selection.

sk1p, 2022-07-28:
Nowadays, we also have CUDA/cupy support, which needs to be taken into account when selecting the right FFT functions - for UDFs, we could offer a similar API as `UDF.xp` to expose the FFT module; maybe `UDF.fft`?