LiberTEM/LiberTEM #413: docs: sphinx-autobuild rebuilds multiple times for each change

sk1p, 2019-09-02:
Something triggers multiple builds after a single write. Maybe this is related to my editor (vim), as it doesn't seem to happen if I just `touch` the file.

uellue, 2019-09-02:
Yes, I am observing the same. Probably it has to re-build internal indices in a first run, and then build the  final version? With LaTeX, up to three consecutive builds are necessary to stabilize a document: First run, build indices. Second run, render indices in document. Third run, update page numbers in references within the document.

sk1p, 2019-09-02:
> Probably it has to re-build internal indices in a first run, and then build the final version

Don't think this is the case here; for example, when [deploying the docs](https://travis-ci.org/LiberTEM/LiberTEM/jobs/578853215), they are only built once; also, when using touch to just change the `mtime`, only one pass is done. I suspect this is because of the way editors save files; one way to debug this is to have a look at the raw inotify events caused by saving a file. I guess I'll do this later, when I'm sufficiently annoyed by this bug :grin:

uellue, 2022-07-28:
Seems to be fixed in current releases -- only one build is performed when I touch a file.

closed 2022-07-28 by uellue