LiberTEM/LiberTEM #431: Content of .tar.gz asset on GitHub, PyPi and Zenodo

uellue, 2019-09-18:
The .tar.gz asset on GitHub contains a dump of the entire GitHub repository, while the assets on PyPi and Zenodo contain only the source code of the Python module, as far as I can tell.

Is this intentional? Perhaps the upload on Zenodo could contain the full repository as well since that would be required to re-create the state of the project should GitHub go under in the far, far future?

uellue, 2019-09-18:
If at all possible, we could dump Issue tracker etc as well? :-)

sk1p, 2019-10-15:
> The .tar.gz asset on GitHub contains a dump of the entire GitHub repository, while the assets on PyPi and Zenodo contain only the source code of the Python module, as far as I can tell.

PyPI: intentional, as `sdist` is a standard Python format which you can directly install.

Zenodo: agree, should be changed! It could include: wheel, sdist, repo archive. Could be created using `git archive` for example. Or we could download from the GitHub releases thingy via https://github.com/LiberTEM/LiberTEM/archive/{tag}.tar.gz (pro: same files for both, con: creates an ordering between github release and zenodo upload). Should we do this for the next feature release? 

> If at all possible, we could document Issue tracker etc as well on Zenodo? :-)

Yeah, from my experience, I always look for some GitHub badge, for example on documentation sites, to check out the issue tracker, license etc. - is it possible at all to include images in the zenodo README? I would suggest we create a small HTML preamble that includes the most important links and prepend that to the rendered README HTML.

uellue, 2019-10-15:
> Or we could download from the GitHub releases thingy via [https://github.com/LiberTEM/LiberTEM/archive/{tag}.tar.gz](https://github.com/LiberTEM/LiberTEM/archive/%7Btag%7D.tar.gz) (pro: same files [...]

Downloading from GitHub sounds good enough for me. :-)

> I would suggest we create a small HTML preamble that includes the most important links and prepend that to the rendered README HTML.

See PR #455!

uellue, 2019-10-15:
> Should we do this for the next feature release?

Hm, is there a reason to not do it for the next point release?

sk1p, 2020-02-10:
> Hm, is there a reason to not do it for the next point release?

Let's try to do this for 0.4.0.

PR #583 (2020-02-10): Upload release repo snapshot to zenodo
Download repo snapshot from `/archive/{tag}.tar.gz`, upload that to zenodo and move zenodo upload after GitHub release creation. Fixes #431.
commit 2b8f64c3 (2020-02-10): Upload repo archive to zenodo, fixes #431
closed 2020-02-10 by sk1p