LiberTEM/LiberTEM #435: Check and optionally update default tileshape behavior for BLO files

uellue, 2019-09-23:
How to reproduce:

Open a BLO file that is not in the parameter cache, for example the Glasgow file.

Observed behavior: The default tile shape is 1,8,128,128. However, the signal dimension of this dataset is (144,144), which makes (128,128) seem like an odd choice.

Question: Is this sensible, is the tileshape parameter still necessary at all for BLO, and/or can we guess better defaults? Perhaps something like (1, 6, 144, 144)?

Since BLO files tend to be small, this is not a big issue in practice at the moment.

sk1p, 2019-10-07:
Hm. This is kind-of related to #247 - ideally the parameter would not exist (maybe we can remove it for 0.3?) I would prefer to have a global (or maybe even per-node) setting "default block size" or similar, which guides the setting of the tileshape. It could default to be initialized from CPU cache size, if we can determine it reliably (Maybe psutil provides something? Could not find anything on a first glance).

uellue, 2020-04-09:
Refs #706

PR #706 (2020-04-27): WIP: Tiled reading in more file formats
Fixes #554, fixes #634, fixes  #27, fixes #331, fixes #373, fixes #635, fixes #435, fixes #247 

## Functionality implementation

- [x] Update `MemoryDataSet` #634 
- [x] Create API for tileshape negotiation #554 
- [x] Add tiling scheme and current index to `UDFMeta` #635 
- [x] Update all `DataSet` impls to work with the new infrastructure
  - [x] hdf5 - can't use the I/O backend as I/O is done by HDF5 itself
  - [x] ser - I/O is done by ncempy, so we can't use the I/O backend
  - [x] raw
  - [x] mib
  - [x] blo
  - [x] k2is
  - [x] frms6
  - [x] empad [... cut]
closed 2020-04-27 by sk1p