LiberTEM/LiberTEM #436: Bounds checking for shapes in GUI

uellue, 2019-09-23:
How to reproduce:

Open a file and add an analysis that has GUI control elements. Use keyboard controls to adjust the size.

Observed behavior:

1. When reducing an inner radius, the value doesn't reach zero, but starts cycling between an arbitrary x and (stepsize-x).
2. The outer radius can be increased without apparent limit

Expected behavior:

1. The inner radius goes to zero with the last step that is smaller than the step size and remains zero when trying to reduce it further
2. The outer radius is limited by the diagonal of the control.

sayandip18, 2020-04-08:
I have managed to reproduce this. What are the prerequisites for this issue?

sk1p, 2020-04-16:
@sayandip18 this is mostly an issue that can be solved if you know a bit of TypeScript. The draggable handles all have some kinds of constraints applied already, so this issue should be fixable by changing those constraints.

Related: on a data set that is has a 1 for one of the navigation axes, the frame picking mode errs out - I think the default values need to be fixed there.