LiberTEM/LiberTEM #440: Opening a HDF5-based Matlab 7.3 file takes a lot of resources

uellue, 2019-09-25:
When opening a Matlab .mat file in format 7.3, which is based on HDF5, one worker is using the CPU for several minutes. The task manager shows moderate file IO activity at the same time. During that time, LiberTEM is not responsive and there's no apparent way to terminate the process except restarting the LiberTEM server. I didn't investigate if that happens with all Matlab files or only specific ones.

[image]

Opening the same file with HDFView 3.1.0 gives a similar behavior.

uellue, 2019-09-25:
See also #186 !

sk1p, 2019-10-07:
Agreed on the expected behavior. Can you find out the parameters of the HDF5 file? I.e. compression, chunk size, (chunked at all?), ...

On another note, the HDF5 reader is not very well optimized yet, there may be some low-hanging fruit there. For example, one may need to explore HDF chunk size vs. libertem tilesize - if the buffering settings of h5py are non-optimal, there may be read-amplification happening (which, in case of compression, means decompressing a chunk multiple times etc.).

uellue, 2019-10-10:
It turns out that this particular file is very, very complex. The output of `h5dump --header` is about 100 MB in size. For comparison, the dump of the EMD version of the EMPAD dataset is only 4 kB. In PR #449 I introduced a timeout which resolves the lock-up for overly complex HDF5 files.

Are there other scenarios where such a lock-up would be relevant?

sk1p, 2019-10-11:
It's kind of sad that HDF5 doesn't have a proper query interface. Running some `SELECT name, size, dtype, shape FROM nodes WHERE node_type = 'dataset' ORDER BY size DESC LIMIT 10` query should not be a problem, even if the database is large. See for example:

https://www.cybertec-postgresql.com/en/postgresql-improving-sort-performance/

"In this case a bit more than 100 MB of data is moved to disk [...] Execution time: 6606.637 ms" - not several minutes, and this is before any optimization! After creating an index: "Execution time: 0.119 ms" - in this specific case, we wouldn't even need the sorting, we could just go with a `MAX` on `size` and be done with it - also with an index this kind of stuff returns almost instantly, or takes a simple table scan without index.

Or one could say: HDF5, used like this, is not the right tool for the job if you want to have a large "schema" and want to discover things about this schema later. Argh. I guess matlab just lazily constructs a tree view in the GUI from this, which should not be a problem.

If there were a way to get the size of a complete group, without iterating, there would at least be a way to heuristically walk the tree.. hm. If HDF5 is used more for 4D-STEM data in the future, we may also have to create a simple treeview for selecting the data set path...

> Are there other scenarios where such a lock-up would be relevant?

and...

> LiberTEM is responsive even if such a long-running operation is underway. The user can terminate it

Hmm, other examples may be: suddenly-slow network file system, opening a too-large file on a laptop, wrong parameters chosen for some analysis etc. - solving these may require a more sophisticated canceling mechanism. I'll need to look into that more in detail.

There are some more catastrophic failure modes, but that's a completely different category (for example, high memory pressure, OOM killer going nuts, system unresponsive etc.)

About this specific problem: did you also observe increasing memory usage or something like that?

uellue, 2019-10-11:
Actually, there's a proprietary query language for HDF5 based on SQL: http://www.hdfql.com/

I am wondering why they release it free of charge but under a proprietary license. The worst of both worlds imo. :shrug:

PR #449 (2019-10-14): Trigger a timeout when guessing parameters for HDF5 takes too long
TODO better error handling. Right now, the opening dialogue defaults to RAW
if the timeout is triggered.

TODO There seems to be code duplication. between _get_datasets() and detect_params().
@sk1p is there a specific reason to do it like that? I have to admit that I don't fully
understand the consequences of changing this.

TODO: Choose a well-balanced timeout value in case traversing a legitimate file
takes longer than that, for example over network.

Closes #440
commit 933f89ad (2019-10-14): Trigger a timeout when guessing parameters for HDF5 takes too long

TODO better error handling. Right now, the opening dialogue defaults to RAW
if the timeout is triggered.

TODO There seems to be code duplication. between _get_datasets() and detect_params().
@sk1p is there a specific reason to do it like that? I have to admit that I don't fully
understand the consequences of changing this.

TODO: Choose a well-balanced timeout value in case traversing a legitimate file
takes longer than that, for example over network.

Closes #440
commit 98360c43 (2019-10-14): Trigger a timeout when guessing parameters for HDF5 takes too long

TODO better error handling. Right now, the opening dialogue defaults to RAW
if the timeout is triggered.

TODO There seems to be code duplication. between _get_datasets() and detect_params().
@sk1p is there a specific reason to do it like that? I have to admit that I don't fully
understand the consequences of changing this.

TODO: Choose a well-balanced timeout value in case traversing a legitimate file
takes longer than that, for example over network.

Closes #440
closed 2019-10-14 by sk1p