LiberTEM/LiberTEM #447: Audit test suite for run time

sk1p, 2019-10-07:
We have some test cases that take several seconds to run. For example, from [a recent CI run](https://travis-ci.org/LiberTEM/LiberTEM/jobs/594576315):

```
5.82s call     tests/test_analysis_masks.py::test_numerics_fail
5.36s call     tests/udf/test_blobfinder.py::test_run_refine_affinematch
4.99s call     tests/udf/test_blobfinder.py::test_run_refine_sparse
4.21s call     tests/test_analysis_masks.py::test_numerics_succeed
3.45s call     tests/server/test_job.py::test_run_job_1_sum
```

That's like 30% of time in the top 5 test cases... :frowning_face: 

This is not only an issue in CI, but also when running the tests locally. We should strive to improve performance here, to encourage devs (us!) to run the tests before committing. Possible to improve CI time: run slower tests in different job in parallel. But that would not improve local test runs, and also has quite some set-up overhead for the extra job. We could opt to have "quick" tests for running locally, which exercise the whole system but omit slower test for special cases.

In my opinion, CI time is still okay-ish, but we should look out to not have it spiral out of control.

Thoughts?

uellue, 2019-10-07:
It would be interesting to know what is taking so long there. Would it be possible to get a profile of the long-running tests in a convenient manner?

In some cases it will be Numba compile time, which is hard to avoid here.

sk1p, 2019-10-08:
In the case of numba, there is a way to [cache the compilation results on disk](https://numba.pydata.org/numba-doc/dev/user/faq.html#there-is-a-delay-when-jit-compiling-a-complicated-function-how-can-i-improve-it). That cache could even be kept for subsequent travis runs. I just wonder if it is possible to enable this conditionally in test runs. Also, caching is hard, so it may be an additional error source...

For profiling, there is [pytest-profiling](https://github.com/manahl/pytest-plugins/tree/master/pytest-profiling) - don't know if it will run with pytest 5.x, though.

uellue, 2019-10-09:
@sk1p I came across this FIXME when reviewing run times: https://github.com/LiberTEM/LiberTEM/blob/ec74c1346d93eff58d9e2201a7ead5af7aa7cf44/tests/conftest.py#L41

I had a shot at it by following https://github.com/pytest-dev/pytest-asyncio/issues/85 and redefining the scope of some built-in pytest_asyncio fixtures. I could get it to work in principle, but the speed-up wasn't great for unknown reasons and a number of tests seem to expect a fresh server.

Regarding the other tests, it was a good idea to review. With the exception of Numba compile times, there was some low-hanging fruit with significant impact. PR to follow!

PR #448 (2019-10-09): Improve slow tests, instructions for profiling, rewrap
Closes #447
commit 9be10926 (2019-10-09): Improve slow tests, instructions for profiling, rewrap

Closes #447
closed 2019-10-09 by sk1p