LiberTEM/LiberTEM #457: Caching layer implementation status

sk1p, 2019-10-15:
I've started to implement #136, see [the ds_caching branch](https://github.com/LiberTEM/LiberTEM/tree/ds_caching). It can right now be considered a prototype. This is supposed to be an overview of what is done and still TODO.

Bigger building blocks:

 - [x] Low-level `WriteHandle` implementation (lives in `libertem.io.writers.base`) that just safely writes tiles to a file (currently using mmap)
 - [x] Distributed Raw DataSet (`DistRawDS`? naming not finalized, currently `libertem.io.dataset.distributed_raw`): writable, single partition per file, structure copied from source dataset, re-uses I/O functionality from our raw file reader
 - [x] `CachDataSet` (`libertem.io.dataset.cached`) using the `DistRawDS` and a source dataset to lazily cache partitions as they are used on worker nodes
 - [ ] Cache eviction / cache statistics
 - [ ] Transformation-style operations that create a distributed raw dataset from a source dataset with a transformation applied (basically #201)
 - [ ] Creating a new `DistRawDS`s from "thin air"; the simulation use case (basics are there, user-friendly interface TODO)
 - [ ] Integration into user-facing APIs
 - [ ] Integration into the web interface (configurable cache path, checkbox "enable cache" when opening, not much more!)

Related and future items, which may influence our implementation: 

 - [ ] Live acquisition (just to have it on the radar, doesn't have to be _done_ as part of this issue)
 - [ ] Create a unified I/O layer that is used by all readers, which then also supports HTTP and HTTP-Based protocols as backend (#278?), and uses liburing (#268) for local SSD I/O

Smaller items: 

- [ ] Cache key from source dataset
- [ ] Tweak some interfaces and names (`DRawDataset -> DistRawDS`? Also create a proper "writable dataset" interface)
- [ ] A lot more testing and coverage, some better error handling and messages (for example: inconsistent state on disks -> should be caught and handled by cache layer transparently)
- [ ] diagnostics for `DistRawDS` and `CachedDataSet`?
- [ ] Maybe fix `get_locations` and `DataSet.initialize` such that they don't need to "know" the `JobExecutor` interface (might not be possible....)
- [ ] Related to `get_locations`, fix the dask-array integration
- [ ] Configurable hostname -> path map, not only for testing but also for non-homogenuous deployments

uellue, 2019-10-15:
As a general remark, we should make it a habit to include version information and changelog info early-on. :-)

uellue, 2019-10-16:
This is a surprisingly small change! 👍 

How does it find the correct cache path for an external location? Will users always have to remember where they cached a DS or is there a way to match external source and cache automatically?

Reading the code, the "processing logic" seems to determine which parts it wants to process next and reads them. Each worker gets their own partition to cache, correct? As far as I can see, the processing stalls for the time that it waits for the tile? That behavior should be fine for simulations and caching. If processing a cached DS is interrupted, an optimization could be to first process cached parts of a DS and then proceed to uncached, but as long as we always process partitions in the same order, that should happen automatically, right?

I am a bit unsure if this structure can handle live data as well or if it will have to be rewritten extensively. For live processing, we will have to change the processing logic. We receive the data in the order that is determined by the experiment, likely filling up one partition at a time. We'll have to process the incoming tiles on all available worker processes on an acquisition node in a round-robin fashion. Completed partitions can transition to offline processing. Here we should make sure to only attempt processing of completed partitions since we can't just make partition data appear as we please, but must follow the flow of the acquisition. In any case, if we can't think of handling this scenario right now, we can figure it out later and rewrite as soon as we have a better handle on this.

uellue, 2019-10-16:
About naming: `ClusterDS` and `CacheDS` to avoid confusion with dask.distributed?

sk1p, 2019-10-16:
> About naming: ClusterDS and CacheDS to avoid confusion with dask.distributed?

Currently, all `DataSet` subclasses use the full form, so that would be `ClusterDataSet` and `CacheDataSet`. As `ClusterDataSet` can also have use-cases on a single-node system, maybe we can find a better name still.

> As a general remark, we should make it a habit to include version information and changelog info early-on. :-)

Agreed and added to the list :grin:

> This is a surprisingly small change! +1

Well, it's not done yet :grin:

> How does it find the correct cache path for an external location? Will users always have to remember where they cached a DS or is there a way to match external source and cache automatically?

That's the "Cache key from source ds" part.

> Reading the code, the "processing logic" seems to determine which parts it wants to process next and reads them

Not sure what you mean by this: a UDF Task asks its associated `Partition` for tiles. It does not control in which order the tiles are read. The `DataSet` controls the order in which `Partition`s are returned, and the `Partition` does the same for tiles.

On the top level, the scheduler of dask-distributed decides the order of execution. From our perspective, all partitions are processed "in parallel", and not in a specific order. This is the part that will be different for live acquisition, for sure.

> Each worker gets their own partition to cache, correct?
> As far as I can see, the processing stalls for the time that it waits for [reading] the tile?

Yes, both correct. A cache miss works like this, for each tile: `[read: I/O wait] -> [processing: CPU bound] -> [write: mmap, page faults]`. Needs to be compared to pipelining the next tile read while processing the first tile, which would also mean we have more data in flight (read: more pressure on CPU caches).

> If processing a cached DS is interrupted,...

... or maybe more common, if previously a ROI was used which didn't cover that partition, ...

> ... an optimization could be to first process cached parts of a DS and then proceed to uncached, but as long as we always process partitions in the same order, that should happen automatically, right?

I don't think we get any determinism/ordering guarantees from dask-distributed... anyways, using a ROI could create "holes" in the cache in arbitrary places, so it would not happen automatically.

Instead of doing "hits first, then misses" (which optimizes for fast feedback rather than total run time), we should instead overlap the cache population with computation on already cached partitions. I'll need to think a bit about this. For example, the partition could indicate whether it is cached or uncached, and be put in different work queues, which are worked on independently. That would guarantee that we always work on filling the cache, as long as there is any work in the uncached queue.

> I am a bit unsure if this structure can handle live data as well or if it will have to be rewritten extensively. 
> We'll have to process the incoming tiles on all available worker processes on an acquisition node in a round-robin fashion.

I think the `ClusterDataSet` should also work for acquisition. On an acquisition node, there would be a single `ClusterDataSetPartition` open at a time, into which a single thread or process writes all incoming tiles.  The processing should be done independently from writing IMHO.

> Completed partitions can transition to offline processing. Here we should make sure to only attempt processing of completed partitions since we can't just make partition data appear as we please, but must follow the flow of the acquisition.

Incomplete partitions are currently saved into a temporary file, and once complete, moved in place atomically. So unless you make an effort to open said temporary file, you don't get incomplete partitions from the `ClusterDataSet`. Of course, what *you write* into the data set can be incomplete...

I think the main difference overall is that we no longer have `Partition`-based parallelism, but that it is finer-grained. There is no longer a "create tasks for all partitions and run them in whatever order" but instead "a constant stream of tiles from partition after partition".

> In any case, if we can't think of handling this scenario right now, we can figure it out later and rewrite as soon as we have a better handle on this.

True.

sk1p, 2019-11-07:
As a high-level API for the caching, there are at least two parts needed:

1) Settings for the cache, i.e. strategy, path, capacity, ...
2) Enabling caching for a specific data set

For 1), I think a settings dict could be passed to the `Context` on initialization, and updated with some other method, like `ctx.update_settings({...})`. What do you think?
For 2), I thought something simple like `ds_cached = ctx.cache(source_ds)` could work.

uellue, 2025-03-26:
Closing since it is better to do this at the OS level.

closed 2025-03-26 by uellue