LiberTEM/LiberTEM #473: Meta issue: Work on executor

uellue, 2019-11-08:
This Issue collects other Issues that depend on extended capabilities in the executor, i.e. where just running Dask futures in the cluster in the current way may not be enough.

- #364 GPU processing could require dedicated worker processes for CPU and GPU processing, and balancing between the two.
- #335 Shared memory requires one function to set it up once, and then others to attach to it while a job runs, and then a reliable mechanism to clean it up. Avoid race conditions, dead locks and orphans.
- #186 Parallel execution of CPU, mass storage and network bound tasks, plus small service tasks like file browsing. That means the OS on the processing node should drive the decision on which task is executed next without oversubscription or idling of the CPU. Frequent context switching is quite expensive since we lose the CPU cache! See also https://unix.stackexchange.com/questions/466722/how-to-change-the-length-of-time-slices-used-by-the-linux-cpu-scheduler
- #242 Per-node intermediate reduction.
- Live data processing, similar to #186

uellue, 2021-01-08:
Support for Dask-based execution should probably remain as an option even after we have an advanced executor, since Dask allows to run on ipyparallel clusters and enables integration with Hyperspy et al, see #922.

uellue, 2021-05-27:
@sk1p @matbryan52 This could be the right issue to coordinate the executor and UDFRunner redesign.

matbryan52, 2021-06-16:
Wondering if either there has been progress/planning on this or, if not, should we make a plan?

I've been examining the architecture and think I have a reasonable idea of how it all fits together - I've been adding to sort of (simplified!) diagram here: https://gist.github.com/matbryan52/21d32163af26fca2ffa521fdfa6c836f

uellue, 2021-06-16:
@matbryan52 Thank you for the diagram! It looks a bit intimidating. :-D It will be a great tool to understand the flow, we can work on refining that and then restructure it for the new requriements.

Regarding plans, the week before we have released LiberTEM 0.7 and almost finished LiberTEM-live 0.1. This week we have instrument time and are doing live processing, with a lot of ad-hoc coding. Next week we should release LiberTEM-live 0.1 and Ptychography 4.0, if things go well.

That means we could make plans for the next steps some time in the second half of next week or the week after. Sounds good? :-)

sk1p, 2021-06-16:
> I've been examining the architecture and think I have a reasonable idea of how it all fits together - I've been adding to sort of (simplified!) diagram here: [...]

Looks good! It's good to see this laid out like this, as a basis for further discussion. I think the main points that will change as part of this work are the `JobExecutor` and `UDFRunner` classes. On the diagram you can already see that the `UDFRunner` is a bit weird and lives both on the main node and on the worker nodes, which should be split up, among other things.

> That means we could make plans for the next steps some time in the second half of next week or the week after. Sounds good? :-)

Sounds good to me!

matbryan52, 2021-06-16:
It's certainly instructive to try to represent the program flow like this, although the layout probably suffers a bit because I'm figuring it out as I go. I do wonder if it would be clearer architecturally if each `UDFTask` spawned some kind of `UDFWorker` rather than an entirely new `UDFRunner` (which could then be renamed something like `UDFDispatcher` I suppose). I don't _think_ any of the methods used in the Task instance of Runner are called directly on the main node...

I'm continuing to update the diagram and will carry on doing so for now. Things like Corrset, IOBackend.

Good luck with the live work and releases, I'll be here to discuss when you're ready.

sk1p, 2021-06-16:
> I'm continuing to update the diagram (at the same link) and will carry on doing so for now. Things like Corrset, IOBackend etc.

Ok, let me know if you have any questions! The `IOBackend` code is a bit hairy, if you like, we can walk through it together.

matbryan52, 2021-06-18:
> The IOBackend code is a bit hairy

It's not so bad! Although I haven't gone too far into how all the syncs, offsets and mmaps are handled as it's probably a bit out of scope for rethinking the Executor.

Major re-factor on the diagram to more clearly split the code running on the master node from that of the Tasks, including the re-creation of UDFRunner and copying of the UDFs - it makes the thing a lot more readable/follow-able. I also forgot to mention that each box on the diagram acts as a URL to the associated definition in the repository.

uellue, 2021-06-30:
Discussions with @sk1p and @matbryan52:

We want to support both live processing and offline processing, in particular on a Dask cluster with Dask arrays. In order to share as much code as possible between the two scenarios, we could consider an abstract interface of "array streams" as the input part for the generic processing. For each data source we would create a "driver" that can create such an array stream and feed it into the generic engine.

uellue, 2021-06-30:
For the current K2 prototype: The UDP receiving, ordering and decoding part could be separated from the actual processing. The decoder would then send the parts to the appropriate recipients to guarantee that complete partitions are processed.

uellue, 2021-06-30:
Should implementation of data assembly for `process_frame()` / `process_partition()` happen in the "driver" or in the "engine"?

## In the driver

- The individual driver processes may not have the data available to do that -- namely K2! --> Would require IPC within the driver!
- Different data sources may share the same issue, namely K2 and Dask arrays that are chunked in the signal dimension. That means the code would be easier to re-use if frame assembly is in the engine.

+ The receiving side could perhaps be simplified, for example by always receiving whole partitions. --> But that might introduce latencies because a whole partition will be buffered: Turn-around is buffering time + transfer time + processing time

-> Doesn't look ideal, probably better in the engine

## In the engine

How is missing data detected or handled? Can we guarantee with enough certainty that all data for a partition will arrive eventually, or should the engine make a decision to skip a tile at some point? --> side channel for buffer management / `damage`, see below.

The "driver" should make a best effort to supply the data of a partition "roughly in order" so that `process_frame()` can run soon and doesn't need too much buffer.

Chunk size: What size would be efficient to decode/transport/process? 1 MB? Also highly implementation specific: Memory mapped partitions more like 100 MB to 1 GB!

--> The engine should be very flexible regarding tiling scheme and tile size for the data it receives: From whole partition (mmap) to "whatever the camera or Dask array creator thinks makes sense"
--> Re-tiling in the engine if the driver likes to send whole partitions, but the UDF prefers smaller tiles? Currently we guarantee that tiles are contiguous, right? --> no simple slicing in signal dimension! How much sense would re-tiling in the engine make, overall? Maybe better to tile correctly in the driver right away?
--> Tile shape negotiation between driver and engine before a run?
--> Engine should probably be able to both process tiles directly (no copy) and to buffer them for frame assembly.
--> Memory ownership: By the engine or the driver? Zero copy means that on a single-node system, the driver fills a buffer which the engine then processes directly if possible (`process_tile()`). As soon as the engine finishes (blocking call?), the driver can re-use the buffer.
--> Since the driver ultimatively decides the tiling and receives the data, probably it should own the memory for the buffers? The engine _might_ keep its own buffer for `process_frame() \ `process_partition()` if needed.
--> Would there be transports, for example network, where buffers owned by the engine make sense, for example 
 to [recv_into()](https://docs.python.org/3/library/socket.html#socket.socket.recv_into)? Should that be handled by the engine or by a sort of adapter that conceptually is still part of the driver?

## What happens if the engine(s) can't keep up?

* We should be robust against temporary bumps -- Numba warm-up, GC, other things happening on the system
* Since the driver will likely own the memory of transfer buffers, it should probably keep enough buffers around to buffer perhaps 100 ms or 1 s, or even more (practice will tell, should be adjustable). After that it probably should throw away contents of older buffers to re-use them for new data. --> Answers question above: Engine _should_ be able to tolerate missing buffers.

Maybe side channel between driver and engine for buffer management, i.e. a way for the driver to communicate which buffers it discarded so that the engine knows to skip them? Sth. like the current `damage`? 

The engine signalling that a buffer is completely processed, for example with a blocking function call, could also be a way to create back-pressure for drivers that can respond to it, for example files, Dask arrays or TCP streams.

## Compatibility with current Executor interface for offline data

Running a Task instantiates a "driver" that reads the data of a partition, an "engine" that runs for only that partition, feeds the driver data into the engine, and yields one result for merging.

## Several partitions per engine run

For optimized processing, one engine and driver should be able to yield several partition results for merging after another to be able to work efficiently with small partitions (responsive) and to re-use task data etc.

How to distribute work between several engines running in parallel on separate computers? Probably "partition by partition", i.e. the driver directing the data stream to the next engine that is free. --> Requires N to M coordination since all the driver processes (might be on separate computers, think K2 with Madmax)  need to agree on which engine receives the data for the next partition.

--> Maybe requires a third logical component, the "dispatcher" which tells the driver processes what to send where.
--> The dispatcher decides what to run where, the engines follow
--> The dispatcher should be able to at least detect stalled or offline engines and throw an error, ideally schedule around it!

The system should also be able to detect if a driver didn't send the data it was expected to send -- wrong target or missing.

* A negotiation "who is next" for each partition might introduce latencies if it happens over the network.
* Use a fixed round-robin. Disadvantage: GPU engines might work at a different pace from CPU engines, heterogeneous clusters!
[... cut ...] ust schedule the next unprocessed partition to an engine directly after it completed a partition. One can refine the scheduling later...
--> Possible solution for live processing: Each driver receives a "wait list" with the engines that should receive the next partitions. As soon as the dispatcher receives a partition result, it sends that engine's "address" to all the drivers to append to their waiting list. That way the drivers know in time where to send data, provided the engines can keep up!

matbryan52, 2021-07-01:
A quick question - what's the opinion/thoughts on 'missing' data from a streaming source (especially via UDP streams like the K2). I appreciate that data might come out-of-order and that's handle-able, but if packets are dropped, or we use a 'buffered' strategy and data arrives at a (fixed length) buffer faster than the workers could possibly consume it, then is the intention to have 'missing' values in the final results ?

If that's not the case, does either having a 'limitless' expanding buffer instead make sense? or even a separate reciever node whose job it is to store 'everything' recieved (to very fast storage, for example) such that any missing values from the real time processing can be filled in later ?

sk1p, 2021-07-05:
> what's the opinion/thoughts on 'missing' data from a streaming source (especially via UDP streams like the K2). I appreciate that data might come out-of-order and that's handle-able, but if packets [...]

I think for the first version, it is intended to have missing values in the result, yes. The idea is to first only run operations that can keep up with the incoming data rate.

For later, I think we had discussions on this: once we save the data on disk also, we can switch to a no-longer-live but "online" processing mode, where we have a writer-reader model and process the data in partitions from disk, and don't lose anything. Dropping frames could still be an option, if having a live view is more important than having complete data. It could also be a combined mode, where an additional node handles filling in the dropped, non-live data points.

(NB: that would also allow to change UDF parameters, or even switch out the UDF completely in the running system)

> If that's not the case, does either having a 'limitless' expanding buffer instead make sense? or even a separate reciever node whose job it is to store 'everything' recieved (to very fast storage, for example) such that any missing values from the real time processing can be filled in later ?

One thing about UDP: it really doesn't give you any guarantees that data is actually sent. Even if a receiver separate receiver node is fast enough to store everything to disk, there still can be missing data. Basically any component in an Ethernet network is allowed to drop packets, from the sending device (I think that would be the digitizer in our case?), to the switch, to the receiving network card. There could be a [bit flip somewhere](https://routingfreak.wordpress.com/2011/03/01/catching-corrupted-ospf-packets/) that makes the checksum fail, and the packet is dropped accordingly (hopefully!) Of course this is not desirable, but I think we should be prepared for this happening, in any case.

uellue, 2021-07-06:
Discussion with @matbryan52 and @sk1p :

* Split the UDFRunner into a more executor-specific part (see https://github.com/LiberTEM/LiberTEM/issues/922#issuecomment-874677705) and a more UDF-specific part that can execute several tasks in sequence and keep state between tasks.

uellue, 2021-07-06:
Discussion with @matbryan52 and @sk1p :

* Split the current `UDFRunner` class into four parts:
  1. Executor-specific aspects of `run_for_dataset()`, such as implementation of [`delayed_for_dataset()`](https://github.com/LiberTEM/LiberTEM/issues/922#issuecomment-874677705), sequence and composition of merge operations, manage buffers
  2. Generic (UDF-specific) aspects of `run_for_dataset()` such as preparing `UDFMeta`, result buffer initialization and preprocessing, damage tracking, setting views correctly, actually calling `UDF.merge()`, creating `UDFResults`,
  3. Executor-specific aspects of `run_for_partition()`, such as buffer management, shared memory, aux data management, execution environment
  4. Generic (UDF-specific) aspects of `run_for_partition()`, such as tile shape negotiation arbitration between the individual UDFs in a pass, inner computation loop that consumes the tile stream and feeds it into the UDFs
* With reference to https://github.com/LiberTEM/LiberTEM/issues/473#issuecomment-871332207, ability to run several tasks (partitions) and be a generator of partial results, or to only do a single one. The latter would be used with `delayed`, for example.

uellue, 2024-05-16:
Skimming through the discussion, it seems that most issues are resolved by using the PipelinedExecutor and the Rust-based receiver code in LiberTEM-live?

Closing as pretty much completed! :champagne:

closed 2024-05-16 by uellue