LiberTEM/LiberTEM #484: Make a docker image

sk1p, 2019-11-14:
https://github.com/LiberTEM/LiberTEM/packages?package_type=Docker

The GitHub "packages" feature is now generally available, which means we can easily publish docker images of LiberTEM directly into the GitHub docker registry. A possible starting point would be the AppImage build script, which could easily be converted to a Dockerfile. Good thing is we can skip the miniconda setup and directly start with the miniconda:alpine image, for example.

sk1p, 2020-01-06:
The `Dockerfile` in #528 contains a commonly used "hack" to enable faster building by first installing all dependencies and then installing LiberTEM itself in a separate layer. That means the dependencies are cached and don't need to be installed on each build. But it also means we need to have a `requirements.txt` file, which I currently manually generate using `pip-compile` from [pip-tools](https://pypi.org/project/pip-tools/). Needs to be run in the CI pipeline to make sure the requirements are always up to date!

sk1p, 2020-01-06:
> Needs to be run in the CI pipeline to make sure the requirements are always up to date!

This is now also implemented.

uellue, 2021-09-06:
Now that we have #1070 and integration in web services becomes more prominent, we could think if publishing a Docker image would make sense. We could analyze what kind of image would help people. I'll put it in 0.9 for discussion, together with #1085

uellue, 2021-09-06:
As a side note, the current image contains a custom ipyparallel version, which we likely don't need in a released Docker image.

sk1p, 2021-09-27:
> As a side note, the current image contains a custom ipyparallel version, which we likely don't need in a released Docker image.

Agreed - the docker image also installs all testing dependencies, which should not be present in a release docker image.

Also, there is a new `ipyparallel` release out that should include our fixes - the custom version is no longer needed. I'll quickly update the `Dockerfile`.

uellue, 2021-09-28:
Hm, so does it make sense to support two independent docker images, one published for "latest release", and one internal for the CI pipeline? Or is it better to base our internal CI image on the external one? In most cases it will share the same base dependencies minus the testing ones, but with some notable exceptions like pinning in the release for incompatibilities.

How do we test a "latest release" public Docker image, by the way? I am a bit wary that installing the "latest master" plus testing requirements over a base image might hide issues with package compatibility in the base image, so it would be advisable to run some simple scripts on a few test files by just copying them onto the image, instead of using a package manager to install test requirements etc. What are your thoughts on that? How often should we validate the public image against latest PyPI and distribution releases? Daily, only with RCs, with each run of the pipeline, when sth changes upstream, ...?

sk1p, 2021-09-28:
> Hm, so does it make sense to support two independent docker images, one published for "latest release", and one internal for the CI pipeline? Or is it better to base our internal CI image on the [...]

Currently, the CI docker image is only used for the "dist" and ipyparallel tests. So whatever is in there is not thoroughly tested, and our tests do not test the actual interface that will be exported by the production docker image, which is `libertem-server` and the web API.

One possibility to test the production image would be to run the integration tests against it, in CI. So everything currently under `tests/server/` could, in theory, be run against an instance of the production container.

For the non-ipyparallel dist tests, we could also use the production image for the scheduler and worker containers. The test requirements are only needed in the `tests` container, where we actually run pytest etc., and ipyparallel is needed in the `tests` container and in the `ipyparallel` engine / controller containers.

So the whole CI process would be:

1) In a new CI stage, before testing, build the `libertem/libertem` image and tag it with the revision, so an image called `libertem/libertem:21fd74df` will be pushed to docker hub. Let's call this new job `docker_build`. We can also build the "dev" image here, which contains both ipyparallel and test requirements, to easily share layers without having to pull the image again in another job. We can figure out what layers we want to share - theoretically we can share most of the layers of the build container - just before installation of LiberTEM itself.

2) The `distributed_tests` job depends on the `docker_build` job (if possible by dependencies, like gitlab supports; otherwise we have to run `docker_build` before all other jobs) and uses both the "dev" and the production image `libertem/libertem:21fd74df`.

3) Add a new test job, which runs the web API integration tests against `libertem/libertem:21fd74df`

4) In the `deploy` stage, if all tests succeeded, pull the `libertem/libertem:21fd74df` image and tag it appropriately, for example as `libertem/libertem:continuous`, `libertem/libertem:latest`, `libertem/libertem:v0.8.1rc1`, `libertem/libertem:stable` etc. (one or more of these tags could be used - depending on the git tag / branch etc.)

> I am a bit wary that installing the "latest master" plus testing requirements over a base image might hide issues with package compatibility in the base image, so it would be advisable to run some [...]

Hm, I would prefer running "real" tests - copying over test files etc. means we won't have pytest, so it will either suck to write tests, get results etc. or we will end up re-inventing pytest as we go along... :(

> How often should we validate the public image against latest PyPI and distribution releases? Daily, only with RCs, with each run of the pipeline, when sth changes upstream, ...?

What exactly do you mean? The production image is static - it doesn't need to be re-validated. Only on a rebuild of the image, it needs to be validated. Would you be happy with the CI approach outlined above?

Btw. I had already started on the "production container" work in a local branch, let me know if you want to work on any of these aspects, and also if you think these are a good idea!

uellue, 2021-09-28:
Ah OK that clears it up! So if I pull `libertem/libertem:v0.8.1` I will always have _exactly_ the same image  and not re-baked with the most recent PyPI releases etc?

The concept sounds great! Running the server and dist tests against a production container would be perfect, as you write it really tests the endpoints we care about. This really mimicks the actual use since someone also might want to fire up LiberTEM workers and the scheduler on a cluster using a Docker image.

Maybe providing a Singularity container that kicks off a `libertem-worker` or `dask-scheduler` would also help with the mess on HPC systems regarding module activation vs Python environments, in the future?

uellue, 2021-09-28:
@shabihsherjeel this Issue is probably also relevant for you?

sk1p, 2021-09-28:
> Ah OK that clears it up! So if I pull `libertem/libertem:v0.8.1` I will always have _exactly_ the same image and not re-baked with the most recent PyPI releases etc?

Yup - good for reproducibility and compatibility, kind of bad for security (i.e.: don't expose containers using the old images to the internet!)

> The concept sounds great! Running the server and dist tests against a production container would be perfect, as you write it really tests the endpoints we care about. This really mimicks the actual [...]

:+1:

> Maybe providing a Singularity container that kicks off a `libertem-worker` or `dask-scheduler` would also help with the mess on HPC systems regarding module activation vs Python environments, in the future?

Yeah - although IIRC you can trivially create a singularity container from an existing docker container, so I don't know if we even have to publish one. That would be something to look into - is there a singularity registry we can use freely, or do we need to use FZJ resources for this?

I don't think there is a need to create separate images for workers / schedulers and `libertem-server` though - they are all available from the same venv anyway, so you can run things like this:

```
docker run --rm -it libertem/libertem:latest /venv/bin/dask-scheduler
docker run --rm -it libertem/libertem:latest /venv/bin/libertem-worker <scheduler-ip>
# etc.
```

The same should be possible with the singularity container IIRC.

uellue, 2021-09-28:
> Yeah - although IIRC you can trivially create a singularity container from an existing docker container, so I don't know if we even have to publish one. That would be something to look into - is [...]

Ok, probably a good time would be if some user or us get annoyed enough with modules to look into Singularity. :-)

> I don't think there is a need to create separate images for workers / schedulers and `libertem-server` though - they are all available from the same venv anyway, so you can run things like this: [...]

Yes, perfect! An easy way to guarantee the same Python environment for all components. :-)

uellue, 2021-09-29:
> Ok, probably a good time would be if some user or us get annoyed enough with modules to look into Singularity. :-)

https://sylabs.io/guides/2.6/user-guide/singularity_and_docker.html

Also, this is how to use GPUs in the image: https://sylabs.io/guides/2.6/user-guide/singularity_and_docker.html

Probably doesn't hurt to just include it in the image since we are using GPUs increasingly often?

PR #1144 (2021-12-06): Docker packaging
Fixes #484 

- Rename Dockerfile for dist tests to Dockerfile.dev
- WIP: scripts/release: add docker_tags command to find suitable tags
- update .dockerignore
- WIP: pipeline job that builds a docker image
- WIP: production docker image

## TODO

- [x] Proper tagging
- [x] Documentation

## Contributor Checklist:

* [x] I have added or updated my entry in [the creators.json file](https://github.com/LiberTEM/LiberTEM/blob/master/packaging/creators.json)
* [x] I have added [a changelog entry](https://github.com/LiberTEM/LiberTEM/tree/master/docs/source/changelog) for my contribu [... cut]
closed 2021-12-06 by sk1p