LiberTEM/LiberTEM #499: Support for single-file DM4 4D datasets

sk1p, 2019-12-05:
In PR #497 we added DM file format support, for now only for stacks of single frames. For the case of single-file 4D data sets, our main use case would be converted K2IS data, but that is sadly saved in a transposed way (maybe also depending on DM version), which makes it very slow to read diffraction images. As a general note, in LiberTEM, our algorithms always try to follow the data layout of the input files for efficiency, so our normal algos don't work unchanged.

As there may be different use cases out there, for either transposed or non-transposed data, this issue is here to gather feedback for potential use cases.

See also: https://github.com/ercius/openNCEM/issues/17 and the discussion in #291

uellue, 2021-10-13:
From communication with Gatan: Apparently there's now an option to save or transform 4D DM4 data in an untransposed layout. More details to be discussed! Support for that should be easy, right?

uellue, 2022-05-05:
Required by Xiaoke Mu and colleagues, bumping to next release.

uellue, 2024-05-16:
Implemented in https://libertem.github.io/LiberTEM/changelog.html#v0-11-0 (open single file C ordered) and https://libertem.github.io/LiberTEM/changelog.html#v0-13-0 (conversion for Fortran ordered)

closed 2024-05-16 by uellue