LiberTEM/LiberTEM #504: Release v0.3

sk1p, 2019-12-09:
I would like to release version 0.3 as soon as possible. We should have a discussion what should be part of it. From my side, this would be at least:

[Currently open PRs tagged with milestone 0.3](https://github.com/LiberTEM/LiberTEM/pulls?q=is%3Aopen+is%3Apr+milestone%3A0.3):

 * [ ] Improve API documentation (#496, #494)
 * [ ] Holography reconstruction UDF (#475)

Important open issues:

 * [ ] UDF API documentation improvements #503 - I would start to work on this, but would also appreciate help, for example with coercing sphinx to output the right parts of the API
 * [ ] Hangs and instability of the CI pipeline (#485, #500) - this one might be tricky but it would be bad to have the CI pipeline hanging for example while doing the release...

I would have liked to support Python 3.8 with this release (#482), but it seems that some dependencies may take a while to release wheels or releases compatible with the new Python version (llvmlite, scikit-image).

Any other important issues that absolutely must be fixed for the next release? Any GUI issues that should be tackled?

woozey, 2019-12-10:
Still working on the Holography UDF. Should finish the changes some time tomorrow.

sk1p, 2019-12-10:
> Should finish the changes some time tomorrow

Sounds good!

uellue, 2019-12-12:
Would it make sense to include the release checklist here as an actual check list?

sk1p, 2019-12-12:
> Would it make sense to include the release checklist here as an actual check list?

Good idea for next time!

uellue, 2019-12-13:
Released! 🎉 😴

closed 2019-12-13 by uellue