LiberTEM/LiberTEM #506: Implement and use type system

uellue, 2019-12-10:
Currently, we use type hinting only sparsely in the API.

In the future, we should pervasively specify types in such a way that they are both machine and human readable.

This helps to make our documentation clear and precise. Additionally, it may help to find issues by using automated checking routines like [mypy](http://mypy-lang.org/). The LiberTEM code base has grown to such a volume that such tests help to maintain coherence across the project.

sk1p, 2020-07-22:
See also: https://glyph.twistedmatrix.com/2020/07/new-duck.html

sk1p, 2022-01-26:
As an update, much of `Context` now does have type hints available - while the UDF interface itself is still a bit patchy. There is now opt-in mypy checking in pre-commit, via the [.mypy-checked file](https://github.com/LiberTEM/LiberTEM/blob/master/.mypy-checked).

uellue, 2024-05-16:
We are using type hints in pretty much all new code. Closing here!

closed 2024-05-16 by uellue