LiberTEM/LiberTEM #512: Citations/References: some references don't have a backlink

sk1p, 2019-12-12:
Some citations don't have a backlink, while others do. This needs to be debugged and possibly fixed (maybe upstream in the bibtex support of sphinx)

sk1p, 2020-02-10:
@uellue do you have an idea about this one? See https://libertem.github.io/LiberTEM/index.html#citations - First item, Gibson1997 is cited from the FEM application docs, but no link back. Is it possible that `.. bibliography::` only supports anchors in the same document?

uellue, 2020-02-11:
@sk1p Yes, back links seem to work only in the same document. The forward links work, though. Currently I don't have a good idea how to fix this. Adding a bibliography per document doesn't work.

I've raised the issue upstream: https://github.com/mcmtroffaes/sphinxcontrib-bibtex/issues/194

uellue, 2020-12-14:
Wontfix for now, larger upstream issue: https://github.com/mcmtroffaes/sphinxcontrib-bibtex/issues/194#issuecomment-743761553