LiberTEM/LiberTEM #523: Add 3D MAGiC (856538) as EU grant on Zenodo

uellue, 2019-12-16:
We should acknowledge 3D MAGiC (856538) https://3dmagic.fz-juelich.de/doku.php

Since this is not in the zenodo.org grant database yet, I'll add an acknowledgment to the homepage and include the info to `packaging/zenodo-upload.json` as a note for the time being.

We should check again in a month or two if the grant is available in the grant database on Zenodo.org and then add it to `packaging/zenodo-upload.json` properly.

sk1p, 2020-06-03:
I'll add this to the 0.6 milestone so we don't forget to check before the next release.

sk1p, 2021-01-27:
I checked this again, and it seems to be still missing (from both production and sandbox deployments of Zenodo). Zenodo [claims to use OpenAIRE](https://help.zenodo.org/), but there, the project is available: https://explore.openaire.eu/search/project?projectId=corda__h2020::e9dfaa77e81ed8df2228180342c4185f

I was curious about the procedures involved, and it seems the OpenAIRE project recently changed their export format (https://github.com/zenodo/zenodo/issues/2071), and the 3D MAGiC grant is contained in the new [json format dump](https://zenodo.org/record/4201546). So I guess we have to wait until their new import procedures are deployed.

I'll move the milestone to 0.7, but before that, we can also create an updated deposition for 0.6 with the grant included, I think.

alessiabardi, 2021-02-08:
Another missing projects in Zenodo that was reported by a user of OpenAIRE is https://explore.openaire.eu/search/project?projectId=rcuk________::77c00c370d9d15cba48e4631b70a5bdf (RCUK "Practical applications of IIIF as a building block towards a digital National Collection")

uellue, 2021-02-12:
I've sent a support request to Zenodo about the two grants, let's see!

uellue, 2021-05-27:
The grants have been added. :-)

PR #1033 (2021-05-27): Include 3D MAGiC
Closes #523

## Contributor Checklist:

* [x] I have added or updated my entry in [the creators.json file](https://github.com/LiberTEM/LiberTEM/blob/master/packaging/creators.json)
* [N/A] I have added [a changelog entry](https://github.com/LiberTEM/LiberTEM/tree/master/docs/source/changelog) for my contribution
* [x] I have added/updated documentation for all user-facing changes
* [N/A] I have added/updated test cases
* [N/A] I have included the [rebuilt production build of the client](https://libertem.github.io/LiberTEM/contributing.html?#building-the-client) (only if
change [... cut]
commit e5ad6599 (2021-05-27): Include 3D MAGiC

Closes #523
closed 2021-05-27 by uellue