LiberTEM/LiberTEM #541: Codeclimate observations and possible improvements

sk1p, 2020-01-08:
* More features of the GitHub integration can be used, for example issues as github inline comments, the issue summary comment etc.
 * Some parameters need to be fixed/adjusted
   * for example, redux reducers naturally have many `return` statements, which elsewhere is a code smell
 * Already caught an issue that somehow was missed by `tslint` in vscode :+1:

uellue, 2020-01-20:
Checks for duplication and cyclomatic complexity are very valuable!

The limit on the number of function parameters feels tight when modifying existing code. In Python, adding a new parameter with a default value is "minimally invasive" since it doesn't require any change in downstream code.  A refactor to reduce the number of parameters would usually involve a major change of the API or even architecture.

Would it be possible to impose tighter limits on introducing *new* functions with many parameters and relax limits for modification of *existing* functions? Alternatively, we should just relax the limit in general since it produces many hits where changes would open a can of worms.

sk1p, 2020-01-20:
I don't think it is possible to differentiate new/existing functions - if you look at a PR that moves functions around, many times issues appear both as fixed and new. I would then suggest we relax the limit in general. If you'd like to have a go at it, this is done in the `.codeclimate.yml`, by setting a threshold for [`argument-count`](https://docs.codeclimate.com/docs/maintainability#section-checks). Not sure what to set the threshold to, though...

closed 2022-07-28 by uellue