LiberTEM/LiberTEM #543: Prepare for GSoC 2020

sk1p, 2020-01-09:
We should prepare for GSoC 2020 soon - registration of projects under the PSF umbrella [closes soon (2nd of Feb)](https://python-gsoc.org/index.html#apply). We should review and update [our ideas page](https://libertem.github.io/LiberTEM/gsoc.html). Let's also collect new ideas in this issue.

sk1p, 2020-01-09:
First review of the ideas page:

> To write your application, Mailman gives a few general ideas. 

Maybe better: "To get an inspiration on how to write your application, [...]"

> For Python GSoC it is a general requirement to already contribute a pull request to a sub-organization before submitting a project idea. Please contact us if you’d like some help with that! Improving our examples in code, description and presentation would be both relatively easy and quite useful for us. The examples are currently lagging a bit behind recent changes in the API. You could hunt down discrepancies and suggest updates. Please contact us for the corresponding data to run the examples!

Hm, is this still the case? On last release I think our examples were all in a runnable form, at least. Need to replace this with updated guidance, another pointer to the "good first issue" list maybe.

Ideas list:

> 1) Beginner/Intermediate/Advanced: Implement new analysis workflows or improve an existing one.
We have a number of them from easy to hard on our waiting list. This can give you experience with the product development, design and application side of software engineering, and applied data science. A major part of the work is first figuring out what to implement together with our users, and then how to implement it. You can decide how far you take it: A detailed requirements document, a technical specification, a prototype, or a full production-grade implementation? All of that is useful for us.

IIRC, this was always the first question: what _concrete_ analysis workflows are we missing? As the students are not electron microscopists, maybe we should instead have a list of concrete analyses here.

Maybe having another go at [implementing RDF mapping](https://www.ncbi.nlm.nih.gov/pubmed/27236215)? We could involve people from KIT for this as co-mentors.

> 2) Beginner/Intermediate/Advanced: Compression survey.
Analyze high-throughput compression techniques, dive into lz4/zstd, blosc etc., compare against existing file formats. With this project you can improve your low-level programming experience: Instruction sets, CPU caches, optimizing compilers, auto-vectorization, and so on. Our favorite technology to do work in this area with Python is numba. Can be done basically independent of the LiberTEM codebase. For a beginner project you can compare existing implementations of common compression algorithms for our kind of data. For an advanced project you could test autoencoders.

We could highlight the different projects with different difficulty: easiest is just: write prototype code which compares a large range of compression algos/impls without integration into LiberTEM. Intermediate would involve integrating an existing compression impl into the LiberTEM caching layer, and advanced would mean experimenting with their own ideas for compression, for example [bit packing](https://lemire.me/blog/2012/03/06/how-fast-is-bit-packing/), and comparing that to off-the-shelf solutions like [cat4py](https://github.com/Blosc/cat4py).

I'm still hyped about the bitpacking idea btw. - it should fare very well on our typical data, while showing very good compression/decompression speeds. Maybe we could code a quick prototype to concretely explore how much potential this has?

> 4) Intermediate: Editor for masks.
> Domain knowledge: -

The student should have basic computer graphics knowledge

> 6) Intermediate/Advanced: Cloud caching layer.

"done" and can be removed - integration into GUI and user APIs is still TBD but not a good fit for a project IMHO.

> 8) Advanced: Live visualization of large binary data.

Need to discuss this internally.

sk1p, 2020-01-09:
@ReimarBauer I guess the organizational side of things is like last year, or do you know of significant changes?

ReimarBauer, 2020-01-09:
see here also deadlines. We have a few days left. 
https://python-gsoc.org/index.html#contact

sk1p, 2020-01-28:
What is the status on this? Should we already do a PR to https://github.com/python-gsoc/python-gsoc.github.io/pulls? Was there a response from the admins yet?

uellue, 2020-02-04:
Closing here since https://github.com/python-gsoc/python-gsoc.github.io/pull/110 is merged and the ideas page is complete for now. :-) We can create new issues for specific items as needed.

closed 2020-02-04 by uellue