LiberTEM/LiberTEM #546: Improve control over number of threads for numerics libraries

uellue, 2020-01-10:
Currently, a script or process that uses LiberTEM should set the thread count to one for numerics libraries by setting appropriate environment variables before loading these libraries, as demonstrated in https://libertem.github.io/LiberTEM/api.html#basic-example

The current approach has a number of disadvantages:
* If LiberTEM is embedded in a larger application, this may have to happen very, very early in the start-up process of this application.
* A Jupyter notebook or embedding application may want to use threads for these numerics libraries when it is using them directly and not through LiberTEM
* The LiberTEM control node may benefit from threading in a cluster set-up.
* Depending on the processing, it may be efficient to mix threading and multiprocessing on the workers. In particular, this can use the L3 cache more efficiently if the workload is dominated by processing tasks that are using the libraries that are affected by the threading settings. That includes a wide range of linear algebra tasks, such as the solvers for grid matching.

In the future, the number of threads for relevant numerics libraries on the workers should be controllable independent from the number of threads on the control process or embedding application.

Ideally, a UDF should be able to signal how many threads it would like to use, and the back-end should adapt the setting based on that.

sk1p, 2020-01-13:
This can be quite challenging, as it depends on the numerics libraries used as a backend (see also [this numpy discussion](https://github.com/numpy/numpy/issues/11826)). Besides the environment variable, most of these support some kind of API call to control the threading. Let's try to compile a list of these here:

 * OpenBLAS: you can call a C library function called [openblas_set_num_threads](https://github.com/xianyi/OpenBLAS/blob/develop/USAGE.md)
 * Intel MKL: uses both OMP and mkl-specific mechanisms to control number of threads (both using function calls and env vars), [with mkl specifics taking precedence over OpenMP](https://software.intel.com/en-us/mkl-macos-developer-guide-techniques-to-set-the-number-of-threads)
 * PyFFTW: number of threads can be passed [as an argument to fft functions](https://hgomersall.github.io/pyFFTW/pyfftw/interfaces/interfaces.html#additional-arguments), or via the `pyfftw.config.NUM_THREADS` configuration.
 * numpy fft? I think it is single threaded only
 * numba? depends on threading backend?

Anything I'm forgetting?

I found a partial solution in https://github.com/joblib/threadpoolctl - it can control BLAS backends OpenBLAS/BLIS/MKL - I don't think it has support for `numba` or `pyfftw`.

As a common requirement to any solution, the executor and UDF runner need to be made aware of the threading configuration, in detail:

 * The UDF interface needs a new configuration method (to keep the UDF namespace clean, maybe a more generic configuration method, which currently only supports number of threads? Can also think about previous discussions, like input dtype preferences, or signaling GPU usage)
 * The UDF runner (`run_for_partition`) needs to take care of the local threading configuration - for example using `threadpoolctl`
 * The executor needs to know about the threading config, to create a more detailed schedule for each worker node (i.e. only schedule to a subset of worker *processes* for each worker *node*) - 
 `UDFRunner.run_for_dataset` can forward this information to the executor via `run_tasks` 

Should the configuration method on UDF also have access to any system configuration information? I.e. a [`WorkerSet`](https://github.com/LiberTEM/LiberTEM/blob/76b2326e865fabee35c96e6898d55078220007c5/src/libertem/executor/scheduler.py#L4)? Maybe in addition need number of cores for each host (if there are less worker processes started than cores are available, or something like that?) If possible, we should try to get away with _not_ providing this information, as it makes the interface easier.

Comprehensive documentation should also be written as part of this issue.

> The LiberTEM control node may benefit from threading in a cluster set-up.

This is already _possible_ via the coarse environment variable controls (i.e. don't set any env vars in your client code, only when starting the `dask-worker`s), but mostly this would concern the top-level calls of the `merge` functions, right? So `run_for_dataset` would set a different threading preference for `merge` calls than for the `process_*` calls in `run_for_partition`.

> In particular, this can use the L3 cache more efficiently if the workload is dominated by processing tasks that are using the libraries that are affected by the threading settings. That includes a wide range of linear algebra tasks, such as the solvers for grid matching.

True, especially could be nice for non-tiled operations on larger frames, like halving the number of processes (and thus the number of frames that are in flight concurrently) and using two threads internally in `process_frame` (or even 4, depending on hyper threading gains).

kunal12298, 2020-03-03:
Hello, I am Kunal Gupta and am looking forward to contribute for this project. How should I get started?

sk1p, 2020-03-03:
Hi @kunal12298!

To get started with LiberTEM, the best way is to pick one or more [issues marked with "good first issue"](https://github.com/liberTEM/LiberTEM/issues?q=is%3Aissue+is%3Aopen+sort%3Aupdated-desc+label%3A%22good+first+issue%22) and submit PR(s) for them. For this project specifically, it is also a good idea to read and understand the code of `libertem.udf.base` - that is probably where most of the changes regarding threading decisions will be made. Please also carefully read the discussion in this issue - it contains a lot of information on the problem and possible solutions.

Let us know if you have trouble understanding the discussion above, or if you have any more questions!

kunal12298, 2020-03-05:
Thank you @uellue, and wanted to ask if there is any other issue related to the issue #546

uellue, 2020-03-18:
Temporarily setting the thread count to 1 on the workers while running for a partition is supported in #680. 

Multi-threaded UDFs are a bit trickier at the moment since building reasonable worker maps can be fiddly: Supported number of threads in the UDF vs. number of cores on the actual machine; good balance between multiprocessing and threading; heterogeneous clusters. Furthermore, we often process rather small data portions where the overhead of multithreading can be quite significant.

I'd propose to close here now and revisit multithreaded UDFs as soon as we have a real use case. Shared memory #335 will likely be easier and more effective.

sk1p, 2020-03-18:
> Multi-threaded UDFs are a bit trickier at the moment since building reasonable worker maps can be fiddly: Supported number of threads in the UDF vs. number of cores on the actual machine; good [...]

Yeah, that sounds like something the "advanced executor" should take care of, locally on each node...

There is still one place that is setting the number of threads via env vars: 

https://github.com/LiberTEM/LiberTEM/blob/master/src/libertem/web/cli.py

Do you want to make a PR, or should I take care of it?

closed 2020-03-18 by sk1p

uellue, 2020-03-18:
> There is still one place that is setting the number of threads via env vars:
> 
> https://github.com/LiberTEM/LiberTEM/blob/master/src/libertem/web/cli.py
> 
> Do you want to make a PR, or should I take care of it?

PR coming momentarily!