LiberTEM/LiberTEM #553: Get coordinates in navigation dimension for UDFs

uellue, 2020-01-20:
Currently, `self.meta.slice` gives access to the coordinates of a tile or frame in flattened navigation space with ROI applied. This is useful to assign data to the correct buffer.

In some applications, namely ptychography, the position of a frame in the actual navigation space without ROI applied is important. We should have a function or interface that gives access to this information.

uellue, 2020-01-23:
The physical coordinates should be derived from the coordinates in navigation space in a separate step in my opinion. Currently, a user would have to implement their own routines for this. In the future we will hopefully have nice facilities for this. #121 

IMO we should follow the dataset shape as returned by `dataset.shape.{sig,nav}` to be consistent. Converting this to a different coordinate system would be something that requires "user knowledge".

uellue, 2020-01-23:
> Currently, this is done on a per-dataset implementation basis, i.e. you can specify `scan_size` in the raw dataset - maybe it should be made more generic, see for example #442

That's an interesting idea, implementing this as a general dataset capability instead of being format-specific! :+1:

AnandBaburajan, 2020-08-03:
> the position of a frame in the actual navigation space without ROI applied is important
> there should be a list of coordinates that correspond to the frames that are part of the current tile

Doubt: for example, in case of a `SEQ` file (which is sometimes loaded as 4D), should 'list of coordinates of frames in a tile in actual navigation space' return the actual 3D coordinates or the loaded dataset's 4D coordinates? Thanks.

sk1p, 2020-08-03:
> Doubt: for example, in case of a `SEQ` file loaded as 4D, should the list of coordinates return the actual 3D frame coordinates or the coordinates in 4D? Thanks.

It should be in 4D coordinates - the idea is to not only make the flattened internal coordinates available, which are useful for easy processing, but also the application-specific coordinates / the result of the reshaping, in this case 4D

AnandBaburajan, 2020-08-04:
@sk1p A `64 x 128 x 128` dataset loaded as `8 x 8 x 128 x 128` with `sync_offset = 62` would yield a single tile with only the last 2 frames and the list of coordinates would be `[ (7, 6, 0, 0), (7, 7, 0, 0) ]`.

But if it's loaded as `3 x 5 x 128 x 128` with `sync_offset = 62`, what would be the coordinates of those two frames with respect to the new dataset shape?

sk1p, 2020-08-05:
> A `64 x 128 x 128` dataset loaded as `8 x 8 x 128 x 128` with `sync_offset = 62` would yield a single tile with only the last 2 frames

Okay, I agree so far, but:

> and the list of coordinates would be `[ (7, 6, 0, 0), (7, 7, 0, 0) ]`.

I don't think this is correct - with `sync_offset=-62` this would be the case, as there would be 62 "empty frames" inserted at the beginning, and you would get the first two frames of the source data. With a positive `sync_offset`, you would "throw away" the first 62 frames of the source data, and start at `(0, 0)` with the remaining two frames; the coordinate array would be `[ (0, 0), (0, 1) ]`.

So in your `3x5` case, the two frames would also start at `(0, 0)`... basically, this would be a `np.unravel_index`-like operation on the flattened coordinates, taking `roi` into account.

Does this make sense, or did I misunderstand something?

AnandBaburajan, 2020-08-05:
> the remaining two frames; the coordinate array would be `[ (0, 0), (0, 1) ]`.

Oh, got it. I thought the coordinates should be the actual position of the frames in the original dataset. It's clear to me now, thanks!

PR #793 (2020-12-14): sync_offset, reshape datasets and get coords for UDFs
## Google Summer of Code '20 project

Fixes #256, fixes #553 and fixes #726 and

* The Python API and GUI now allows specifying `nav_shape` and `sig_shape` parameters to set a different shape
* `scan_size` and `detector_size` parameters are deprecated in favor of `nav_shape` and `sig_shape` respectively
* All datasets handle missing data gracefully
* The Python API and GUI now allow specifying a `sync_offset` to handle synchronization/acquisition problems
* Users can now access the coordinates of a tile/partition slice through UDFMeta.coord [... cut]
closed 2020-12-14 by sk1p