LiberTEM/LiberTEM #554: Tile size and read mode negotiation for UDFs

uellue, 2020-01-20:
In relation to #549 and #551: The optimal tile size for `process_tile()` depends on the circumstances:

* The UDF creates intermediate data similar in size to the tile (log scaling etc), the dataset doesn't use mmap or the data is converted before passing to the UDF: Tiles of approx. 1 MB are optimal.
* The dataset can return mmapped data directly without any conversion and the UDF works efficiently on large tiles, for example with a pure dot product with a shallow mask stack: Very large tiles up to partition size can be optimal.

For that reason, the UDF interface should allow for negotiation of the tile size.

Idea by @sk1p: Instead of directly using `get_tiles()` with loads of parameters, the partition could allow to create a ProtoTileStream object which handles the decision logic on Dataset side (full frame? mmap? dtype?) and can give information on the actual read strategy with the given parameters before tiles are loaded.

The UDF can indicate a preferred upper limit for tiles, which could be 1 MB by default. This is in no way guaranteed since reading full K2 frames will be way above this limit.

The UDFRunner can then decide on a preferred tile size based on the ProtoTileStream, UDF method and UDF preferred size information.

sk1p, 2020-01-20:
I really like this, as it allows us to finally remove the bespoke `tileshape` params and some of the  format-specific `tileshape` calculations... it also means we have a central point where we can tweak the tilesize, possibly allowing to make it larger for machines with huge L3 cache (latest AMD Ryzens have up to 256MB L3, that is about 4MB L3 cache **per core**)

So, instead of `get_tiles` on the `Partition`, we would have `get_tilestream` with the same parameters:

```python
def get_tilestream(self, crop_to=None, full_frames=False, mmap=False, dest_dtype="float32",
                   roi=None, target_size=None):
    # ...
    return TileStream(...)
```

The `TileStream` can then be queried for the concrete reading modes (estimated tile size, mmap enabled?, ...) and parameters can be updated accordingly.

```python
# example usage:
stream = partition.get_tilestream(full_frames=True, roi=roi, dest_dtype=dest_dtype)
stream.set_tilesize(min(stream.get_max_tilesize(), udf.get_the_preferred_max_tile_size()))

# then, either of these forms could be implemented:
tiles = stream.get_tiles()
for tile in tiles:
    pass

for tile in tilestream:
    pass
# in the second form, setting a parameter after starting iteration should raise an exception,
# which is implicitly handled in the first form (only after calling `get_tiles` again will the new parameters take effect)
# the name fits better for the second form, IMHO

# also possible: builder-like parameter passing, which is natural for methods with
# many parameters and default values for most of them:
stream = partition.get_tilestream()
stream.set_roi(roi)
stream.set_dest_dtype(dest_dtype)
if method == 'frame':
    stream.set_fullframe(True)
for tile in stream:
    ...
```

Nice side-effect: a `TileStream` can then also contain information about the concrete tiling scheme that will be used for the given parameters - which can be interesting for constructing buffer structures that mirror the given tiling scheme. 

Remaining questions: 

 * Should there be symbolic values for "whatever fits in L3" or "as large as possible", or other values?
 * What does the UDF need as input to decide its maximum tile size? The mask UDF would be happy with returning "as large as possible" without any input, right? In case of mmap, this would be granted, otherwise, the data set would limit it to L3-sized tiles
 * Does it need to know anything about reading modes (mmap, direct, ...)? I guess this could be some kind of namespace (like `UDFMeta`), which can be filled with items as needed

uellue, 2020-01-21:
Yes, symbolic values would be great, similar to `UDF.USE_NATIVE_DTYPE` in order to eliminate opaque magic numbers. Plus, we might be able to tune them for the system at a later time -- AMD EPYC and such!

Interesting aspects from #247 and #310: Not only the size, but also the shape of a tile is relevant. An interface to indicate the preferred number of frames in a tile could be nice. The ideal for numba-based processing is somewhere around 8 frames to use the L1 and L2 cache efficiently in operations like #398.

Probably a dataset could decide then in what way it would split the individual frames since that doesn't matter that much anymore as long as the tiles contain sequential pieces of memory that are "long enough" (a block size of 128 sequential numbers is more than sufficient in my experience). The dataset should know best what way of splitting frames makes sense and produces nice stretches of sequential access.

Some datasets could be very flexible in their tiling scheme, namely RAW. Others like K2IS can only offer a limited set of tilings. Plus, we might have situations where a dataset can offer mmap only for specific tiling schemes -- imagine K2IS with uint16 or float32 instead of packed uint12!

Perhaps the UDF should then pick the tiling scheme that it finds most suitable?

The process could be then as following:

* UDF indicates preferred tile size (soft), preferred frame count (soft), preferred dtype (hard lower bound) and ability to process tiled data through implementing `process_tile()` or not (hard).
* ProtoTileStream generates offers. If it is very restricted like K2IS, it will return items from a short  list of fixed options, filtered by the hard constraints, with "easy" ones first. If it is flexible like RAW, it will generate one best matching option for the constraints . The UDF then picks one, with default operation of picking the first.

uellue, 2020-01-21:
> * Does it need to know anything about reading modes (mmap, direct, ...)? I guess this could be some kind of namespace (like `UDFMeta`), which can be filled with items as needed

Perhaps the information could be available. The main point here is to decide on an upper bound for tile size if the data has to be converted first, right? It could be sufficient to let the ProtoTileStream apply its own upper constraint on tile size depending on the reading method.

uellue, 2020-01-21:
> I really like this, as it allows us to finally remove the bespoke `tileshape` params and some of the format-specific `tileshape` calculations...

Yes! :-) :-) :-)

uellue, 2020-01-21:
Ah, and we are also getting close to adaptive partition sizes #382!

sk1p, 2020-01-21:
> Some datasets could be very flexible in their tiling scheme, namely RAW.

Should clarify this - when tiling in the signal dimension, even the RAW reader has quite some limits, imposed by the actual I/O method. For example, reading "thin" columns means we a) have high overhead in the number of `read` syscalls (when not using liburing), and b) read amplification because block sizes (`PAGE_SIZE` etc.). The read amplification issue is not so bad I think (if those blocks aren't touched by the CPU but stay in page cache), but large number of syscalls could kill performance quite reliably.

uellue, 2020-02-04:
Results from some tests: UDFs can benefit significantly from optimizing the tile size and shape on a case-by-case basis. ApplyMasksUDF is particularly sensitive in case of deep mask stacks. In https://github.com/LiberTEM/LiberTEM-Examples/blob/master/bullseye-poly.ipynb, going from (1, 4, 250, 250) (1 MB) to (1, 16, 16, 250) (256 kB) improves the feature vector calculation performance by a factor of 2, while the stddev and logsum UDFs are much less affected.

That means tile size and shape negotiation can yield sizable performance gains.

uellue, 2020-03-02:
The new `StdDevUDF` #640 could be a good test bed for the impact of tile shape negotiation. It passes twice over a subset of a tile for the "minibatch" two-pass calculation of sum and sum of variances. If a memory-mapped tile contains so many frames that `n_frames * BLOCKSIZE` doesn't fit the L3 cache, it will likely lose performance dramatically. It could even make sense to restrict the number of frames so that `n_frames * BLOCKSIZE` fits the L2 cache.

Being able to give a size hint for the navigation size of a tile saves the additional complication of implementing blocking in navigation dimension. Instead, this level is then delegated to the LiberTEM back-end. The requirements for the `StdDevUDF` would be to specify a minimum and maximum comfortable nav size for a tile, while the optimal signal size can be decided by the back end.

If the data is converted, the total tile size should fit the L3 cache. If the data is memory-mapped, tiles can have arbitrarily large signal size.

uellue, 2020-03-09:
Discussion with @sk1p: We should check if using native input dtype in StdDevUDF and casting only within `process_tile()` brings a performance benefit once tile shape negotiation works. The values are up-casted to `float64`anyway now for stability reasons #672.

uellue, 2020-03-20:
`scipy.sparse.csr_matrix` doesn't work well at all on large memory-mapped partitions, likely for similar reasons as StdDevUDF. This could be another test case and use case, in particular since the `ApplyMasksUDF` will have to signal a tile shape depending on the current back-end:

* OpenBLAS or Torch for dense masks: No restrictions.
* scipy.sparse: Cache-friendly tile height, might also depend on number of masks

uellue, 2020-04-06:
Discussion with @sk1p:

* We can leave the signal size open as a free variable since this is the "fast" axis and typical optimized algorithms do their own blocking for L1 cache friendliness in this direction.
* The total tile size in bytes only has a maximum, no minimum: Always as large as possible to reduce overhead in LiberTEM!

In conclusion, remaining are nav size min and max, and total size max.

uellue, 2020-04-06:
Discussion with @sk1p: We need diagnostic tools.

* Negotiation inputs and results
* I/O method
* Decoding or not

uellue, 2020-04-06:
Information required for size negotiation:
* `input_dtype`
* `ApplyMasksUDF`: Number of masks, back-end (dense, or what kind of sparse, torch, ...) and mask `dtype`
* Possibly `dtype` of result buffers if they are not determined by `input_dtype` -- for example user-provided `dtype`
* Possibly cache size(s) of the CPU it will run on
* With #364: Run on GPU or CPU, which GPU (available RAM -- anything else?)

uellue, 2020-04-14:
DIscussion with @sk1p: Lower and upper bound for signal size does make sense for `kind='sig'` reductions to make sure that the piece of buffer fits the cache.

PR #706 (2020-04-27): WIP: Tiled reading in more file formats
Fixes #554, fixes #634, fixes  #27, fixes #331, fixes #373, fixes #635, fixes #435, fixes #247 

## Functionality implementation

- [x] Update `MemoryDataSet` #634 
- [x] Create API for tileshape negotiation #554 
- [x] Add tiling scheme and current index to `UDFMeta` #635 
- [x] Update all `DataSet` impls to work with the new infrastructure
  - [x] hdf5 - can't use the I/O backend as I/O is done by HDF5 itself
  - [x] ser - I/O is done by ncempy, so we can't use the I/O backend
  - [x] raw
  - [x] mib
  - [x] blo
  - [x] k2is
  - [x] frms6
  - [x] empad [... cut]
closed 2020-04-27 by sk1p