LiberTEM/LiberTEM #555: Deep integration of ApplyMasksUDF with GEMM implementation

uellue, 2020-01-20:
Discussion with @sk1p 

Currently, the ApplyMasksUDF just uses a GEMM implementation through a simple dot product.

Internally, GEMM implementations offer more features, such as re-using internal copies with transposed data and direct accumulation into a result buffer with data from previous steps for tiled processing. See for example https://docs.scipy.org/doc/scipy/reference/generated/scipy.linalg.blas.dgemm.html for a GEMM function signature.

In the future, the ApplyMasksUDF could make better use of such advanced capabilities. The result would be an efficient multi-scale GEMM implementation where low-level libraries optimize on L1 and L2 cache scale, while the Python part optimizes at L3 cache scale and larger.

sk1p, 2020-01-20:
Related: https://github.com/LiberTEM/LiberTEM/issues/98

uellue, 2020-01-21:
Functions like `scipy.linalg.blas.dgemm` can't integrate into sliced output buffers. The parameter `overwrite_c` in combination with `c=accumulator` and `beta=1` only work if `accumulator` is an unsliced array with Fortran ordering. That means directly integrating into the result buffer of an UDF is not easily feasible with this interface. This functionality would require a NumPy-specific dot product. This might be feasible with Numba. However, it would probably not be easy to increase performance and quality compared to using an existing BLAS implementation and subsequent accumulation using NumPy.

uellue, 2020-01-21:
I am closing this for now since there are currently no NumPy-compatible interfaces for this. It would require a NumPy-compatible GEMM implementation.

closed 2020-01-21 by uellue