LiberTEM/LiberTEM #561: Large raw files on Windows

sk1p, 2020-01-22:
One of the errors from #482 actually looks like a valid bug in LiberTEM (Py38 on Windows):

```
_______________________________ test_large_pick _______________________________
large_raw = <RawFileDataSet of uint16 shape=(100, 100, 1216, 1216)>
lt_ctx = <libertem.api.Context object at 0x10762A48>
    def test_large_pick(large_raw, lt_ctx):
        y, x = large_raw.shape.nav
        dy, dx = large_raw.shape.sig
        analysis = lt_ctx.create_pick_analysis(large_raw, y=y-1, x=x-1)
>       result = lt_ctx.run(analysis)
tests\io\test_raw.py:22: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
.tox\py38\lib\site-packages\libertem\api.py:553: in run
    udf_results = self.run_udf(
.tox\py38\lib\site-packages\libertem\api.py:595: in run_udf
    return UDFRunner(udf).run_for_dataset(dataset, self.executor, roi)
.tox\py38\lib\site-packages\libertem\udf\base.py:701: in run_for_dataset
    for part_results, partition in executor.run_tasks(tasks, cancel_id):
.tox\py38\lib\site-packages\libertem\executor\inline.py:21: in run_tasks
    result = task()
.tox\py38\lib\site-packages\libertem\udf\base.py:600: in __call__
    return UDFRunner(self._udf).run_for_partition(self.partition, self._roi)
.tox\py38\lib\site-packages\libertem\udf\base.py:635: in run_for_partition
    for tile in tiles:
.tox\py38\lib\site-packages\libertem\io\dataset\base.py:679: in get_tiles
    yield from self._get_tiles_with_roi(
.tox\py38\lib\site-packages\libertem\io\dataset\base.py:763: in _get_tiles_with_roi
    with self._fileset as fileset:
.tox\py38\lib\site-packages\libertem\io\dataset\base.py:562: in __enter__
    f.open()
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
self = <libertem.io.dataset.raw.RawFile object at 0x10762310>
    def open(self):
        if self._enable_direct:
            fh = os.open(self._path, os.O_RDONLY | os.O_DIRECT)
            f = open(fh, "rb", buffering=0)
        else:
            f = open(self._path, "rb")
        self._file = f
        if self._enable_mmap:
>           raw_data = mmap.mmap(
                fileno=f.fileno(),
                length=self.num_frames * self._frame_size,
                offset=0,
                access=mmap.ACCESS_READ,
            )
E           OverflowError: Python int too large to convert to C ssize_t
.tox\py38\lib\site-packages\libertem\io\dataset\raw.py:109: OverflowError
---------------------------- Captured stdout setup ----------------------------
```

I think this could be fixed by `mmap`ing not whole files, but actually setting offset/length to only map the current partition. The cause would be Windows using the signed `ssize_t` for `mmap` (I guess from the error message), instead of `size_t` like Linux.

sk1p, 2020-01-22:
Error message is from https://ci.appveyor.com/project/sk1p/libertem/builds/30284586/job/r484lt666ryemsta

uellue, 2020-01-27:
When I tested this on my machine, I couldn't reproduce it with Python 3.8.1. Could this be an upstream issue?

sk1p, 2020-01-27:
Could be a build issue - it is not clear to me if the Python 3.8 on AppVeyor is the official build, or not. Did you test with conda, or plain Python? Looking at the error log, there are a few other errors that stand out as weird, for example this one:

```
[00:03:38] _______________ test_pick_analysis_via_api_3_3d_ds_fail_4[UDF] ________________
[00:03:38] 
[00:03:38] lt_ctx = <libertem.api.Context object at 0x1083A9A0>, TYPE = 'UDF'
[00:03:38] 
[00:03:38]     @pytest.mark.parametrize(
[00:03:38]         'TYPE', ['JOB', 'UDF']
[00:03:38]     )
[00:03:38]     def test_pick_analysis_via_api_3_3d_ds_fail_4(lt_ctx, TYPE):
[00:03:38] >       data = _mk_random(size=(16, 16, 16, 16, 16, 16))
[00:03:38] 
[00:03:38] tests\test_analysis_raw.py:289: 
[00:03:38] _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
[00:03:38] tests\utils.py:38: in _mk_random
[00:03:38]     data = np.random.choice(choice, size=size).astype(dtype)
[00:03:38] _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
[00:03:38] 
[00:03:38] >   ???
[00:03:38] E   MemoryError: Unable to allocate 64.0 MiB for an array with shape (16, 16, 16, 16, 16, 16) and data type int32
[00:03:38] 
[00:03:38] mtrand.pyx:996: MemoryError
```

It should not be a problem to allocate 64MiB; also it works well for the other Python versions.

uellue, 2020-01-27:
I tested with conda. Is this perhaps a 32 bit build? That could explain the issues, perhaps? The 64 MB triggering the MemoryError is perhaps just the last drop that exhausts the available address space or memory?

sk1p, 2020-01-27:
I also thought it could be a 32bit built, but it isn't:

```
[00:00:00] Build started
[00:00:00] ECHO %PYTHON% %PYTHON_VERSION% %PYTHON_ARCH%
[00:00:00] C:\Python38 3.8 64
```

uellue, 2020-01-27:
Hm strange. Perhaps we wait a bit longer if it sorts itself out? :-D

sk1p, 2020-01-28:
> Hm strange. Perhaps we wait a bit longer if it sorts itself out? :-D

:+1: it could very well be a strange build on AppVeyor - I'll investigate to run this also on an Azure Windows Pipeline if this is not too much work.

uellue, 2020-01-31:
I'm closing this since it didn't occur on real-world systems with Py38, doesn't happen at all on Py36 or Py37 and it is likely an issue with the build or upstream.

We can re-open it if necessary if this issue occurs again after the Dask issues with Python 3.8 are resolved #482.

closed 2020-01-31 by uellue