LiberTEM/LiberTEM #574: Assorted HDF5 bugs

sk1p, 2020-02-04:
- [ ] Reading of nD HDF5 files, see #567
- [ ] Debug performance: pickling, general UDF perf, scaling behavior
- [ ] Chunk-aware access patterns, if the dataset is chunked

sk1p, 2020-02-04:
> Chunk-aware access patterns, if the dataset is chunked

If you choose a `tileshape` that is a multiple of `chunks`, this already works, but the user shouldn't have to worry about that. 

> Debug performance

ROI performance should now be on-par with other formats, once #567 is merged.

> Performance DaskJobExecutor compared to InlineExecutor. On my Windows laptop the two are nearly equal, despite InlineJobExecutor using only one core. On Binder, InlineJobExecutor is 5x faster compared to DaskJobExecutor reading from a HDF5 file

If inline performance is faster than with dask, this sometimes suggests serialization overhead, see #335. Was this specifically with applying the huge sparse mask stack? Or consistently all over?

sk1p, 2020-02-04:
For chunked access, see also the [chunk cache settings](http://docs.h5py.org/en/stable/high/file.html#chunk-cache) of h5py. Adjusting them can be helpful in some cases, mostly if there is a mismatch between the chunking of the h5 dataset and the tileshape. Without a cache, there can be read amplification, for example if the chunk doesn't fit HDF5's internal cache and we only read a subset of a chunk at a time...

uellue, 2020-02-04:
> If inline performance is faster than with dask, this sometimes suggests serialization overhead, see #335. Was this specifically with applying the huge sparse mask stack? Or consistently all over?

First of all, it scales OK on moellenstedt. Typical was 6-7x faster on 18x the cores. The large feature vector job doesn't scale too well (2.5x faster on 18x the cores). Here we may have both serialization and L3 cache issues. #335 for the mask could help. One should add that moellenstedt is running with a large background load of low-priority simulations that can mess with the caches.

On my laptop, running it on two cores with Dask is consistently equal or somewhat slower than using the InlineJobExecutor. However, I should add that my system is running the typical Win10 bloat (indexing, virus scanner, ...) plus a full desktop environment, which adds up to 25 % system load. That means I may have a lot of things that interfere with the CPU caches.

In conclusion, I'd take this out here, in particular since RAW files perform the same on Binder. This doesn't look like a file format issue, but a general issue with needing large undisturbed L3 cache to perform well.

sk1p, 2020-02-04:
> In conclusion, I'd take this out here, in particular since RAW files perform the same on Binder. This doesn't look like a file format issue, but a general issue with needing large undisturbed L3 cache to perform well.

:+1:

sk1p, 2020-02-10:
I think the most important issue (reading of nD data) is fixed, I'll move this up to 0.5 for a detailed look at chunking and perf.

uellue, 2020-04-07:
Bumping the remaining parts to 0.6 together with #554.

PR #988 (2021-03-10): Don't assume pure functions in `DaskJobExecutor`
Closes #574 

* Updated `run_tasks`, `run_function`, and `map`

## Contributor Checklist:

* [N/A] I have added or updated my entry in [the creators.json file](https://github.com/LiberTEM/LiberTEM/blob/master/packaging/creators.json)
* [ ] I have added [a changelog entry](https://github.com/LiberTEM/LiberTEM/tree/master/docs/source/changelog) for my contribution
* [N/A] I have added/updated documentation for all user-facing changes
* [N/A] I have added/updated test cases
* [N/A] I have included the [rebuilt production build of the client](https: [... cut]
closed 2021-03-10 by uellue