LiberTEM/LiberTEM #588: Document available test datasets

uellue, 2020-02-12:
Thx @twentyse7en  for the suggestion!

Thx @cophus again for the bullseye data. :-)

- [ ] Examples for a collection of public datasets in LiberTEM-Examples
- [ ] Link to files, document where and how to get data in documentation
- [ ] Base feature documentation on public datasets, such as screenshots

twentyse7en, 2020-02-23:
Sorry I forget to add source of dataset used for updating screenshot as you mentioned.
where exactly should I mention the dataset?

sk1p, 2020-02-23:
> Sorry I forget to add source of dataset used for updating screenshot as you mentioned.
> where exactly should I mention the dataset?

I would suggest adding a proper citation directly in the GUI usage docs at minimum. In addition, we can have a section "public datasets" or similar that has a collection of interesting data.

twentyse7en, 2020-02-23:
That would be great :+1:

twentyse7en, 2020-02-25:
Where are you planning to add new section ?

sk1p, 2020-02-25:
> Where are you planning to add new section ?

Hm, this is not so easy to decide. But as we are planning to do a documentation review soon™ anyways, it could start out as a new top-level item in the TOC (top level means it will be listed as one of the items in the Contents list on the left side). Later, as part of the review, we will re-structure our documentation anyways.

It should be linked from all relevant sections, for example "Loading data" for people curious how things work but not having any data of their own, or before loading data in the "GUI usage" section to allow people to follow along on their own machines.

PR #710 (2020-04-09): Various documentation fixes
* Reference to sample data section, closes #588 
* Update creators.json
* Update PR template, closes #646

## Contributor Checklist:

* [x] I have added myself to [the creators.json file](https://github.com/LiberTEM/LiberTEM/blob/master/packaging/creators.json)
* [x] I have added/updated documentation for all user-facing changes
commit d2875da7 (2020-04-09): Reference sample datasets from within the documentation

Closes #588
closed 2020-04-09 by sk1p