LiberTEM/LiberTEM #589: Mention more debugging and profiling techniques

sk1p, 2020-02-12:
## scalene

Need to try scalene, which might be a better experience than the other profilers:

https://github.com/emeryberger/scalene

Includes low-overhead memory profiling, time spent in C vs time spent in Python, ... - may need to check if this properly tracks numpy allocations etc.

uellue, 2020-02-13:
Scalene looks nice! I'll make sure to try it at the next opportunity.

sayandip18, 2020-03-31:
Hey, can I work on this?

sk1p, 2020-04-01:
> Hey, can I work on this?

Sure. Have you worked with profilers before? Did you already play around a bit with the profiling techniques [from the documentation](https://libertem.github.io/LiberTEM/udf/profiling.html#udf-profiling)? A good exercise is to pick some UDF (can be one of the built-ins), run it, and see where most of the time is spent. Try not to limit yourself to only the UDF code, also see how much is spent in LiberTEM internals.

You can try different techniques to see how they differ, and with scalene, you may be able to track memory usage (this is difficult - try to find out if it can track memory that is allocated by numpy).

The `perf` part is a bit harder - I think there it is best to create a list of good introductory links and not go too much into detail.

Let me know if this is enough information to get you started!

sayandip18, 2020-04-01:
No, I haven't worked with profilers before. And that's the reason I chose this issue. 

> 
> 
> also see how much is spent in LiberTEM internals.

What are "internals"?
  
> 
> Let me know if this is enough information to get you started!

I hope so!

sk1p, 2020-04-01:
> What are "internals"?

From a user's perspective, there is a difference between their own code in a UDF (which is what they may be interested in profiling!) and our "internal" code: I/O, "data logistics", general bookkeeping etc.

sk1p, 2020-04-16:
Another profiler: https://github.com/benfred/py-spy - can "attach" to running python processes, including support for subprocesses etc. - looks very useful!

sk1p, 2020-04-19:
For the `perf`-related section: https://github.com/iovisor/bcc/blob/master/tools/lib/ucalls_example.txt

sayandip18, 2020-04-25:
How about this https://github.com/vpelletier/pprofile ? Offers line-granularity and thread-aware profiling.

sk1p, 2020-04-26:
> What about this https://github.com/vpelletier/pprofile ? Offers line-granularity and thread-aware profiling.

You can try and see how well it works - I don't think we are using threads in any performance-critical part, but if the profiler is provably better than the line profiler, we should also include it in the docs.

sk1p, 2020-05-12:
Another one: https://functiontrace.com/

sk1p, 2020-06-03:
Firefox profiler can load `perf` output:

https://profiler.firefox.com/

https://profiler.firefox.com/docs/#/./guide-perf-profiling