LiberTEM/LiberTEM #591: Update screenshots in usage documentation

uellue, 2020-02-13:
The screenshots in https://libertem.github.io/LiberTEM/usage.html are from an ancient release and the GUI offers more and different features now.

See also #588

twentyse7en, 2020-02-14:
I would like to work on this.

twentyse7en, 2020-02-14:
What exactly should I do ? Update the screenshot with the same example for latest version.

sk1p, 2020-02-14:
> What exactly should I do ? Update the screenshot with the same example for latest version.

Yes, that's the first part of the issue, the second part would be to write a description for the ROI and stddev GUI features, including screenshots. The first part is easy - for the second part, you will need to really understand what is going on. Don't hesitate to ask questions!

twentyse7en, 2020-02-14:
For the first part , where can I get the data used in that tutorial ?

For the second part , I would like to know what is going on.

uellue, 2020-02-14:
Perhaps you can use the Zenodo dataset from @cophus and team instead: https://zenodo.org/record/3592520

Use the "circular", not the "bullseye" dataset. Each of the two HDF5 files is a container with several datasets. Best use "4DSTEM_experiment/data/datacubes/polyAu_4DSTEM/data" since it looks a bit more like a real sample than the other. You can use [HDF5 view](https://www.hdfgroup.org/downloads/hdfview/) to explore the files if you like.

If you mention the source of the dataset in the text, you already finished half of #588 as well. :-)

The second part is pretty straightforward, too. You'll see the additional drop-downs below the left display in the current GUI compared to the old screenshots. In the current version you can choose to view the standard deviation or the sum of the frame, and you can calculate sum and stddev for a limited region of interest on the right side. That's a nice addition to pick mode. Best to play a bit with it first, then we can explain more what it does if you like. :-)

twentyse7en, 2020-02-18:
Hey @uellue , completed the first part. I did work on those new option.This is what I understand, ROI is introduced.By default it's `all`. New options are `disk` and `rect`. Regions could be moved and resized. We can choose std and average for those region.

When I was switching between `average` and `stddev` image on left was changing. Could you explain whats actually happening when we select stddev.

sk1p, 2020-02-18:
> When I was switching between average and stddev image on left was changing. Could you explain whats actually happening when we select stddev.

Selecting `average` means we calculate the average image over the selected ROI. If there is no ROI selected, the average is calculated over all "images" in the dataset. If you select `disk`, for example, all images that lie inside the selected `disk`.

If you select `stddev`, instead of calculating the average, the standard deviation over all selected images is calculated.

Do you already understand the nature of these 4D datasets? Have a look at https://libertem.github.io/LiberTEM/concepts.html and let me know if you have further questions!

twentyse7en, 2020-03-07:
I will do the second part too :smiley:

uellue, 2020-04-07:
Closing, implemented with #669. :-) Thx again @twentyse7en!

closed 2020-04-07 by uellue