LiberTEM/LiberTEM #597: Error message when closing dataset

sk1p, 2020-02-13:
Spurious error when closing dataset (SER in this case):

```
[2020-02-13 15:51:28,533] INFO [libertem.web.base.log_message:24] message: FINISH_JOB (job=fb1a2e82-1d4a-4ce0-80a5-624dbcf2b0e0)
tornado.application - ERROR - Uncaught exception DELETE /api/datasets/3ac211e9-4d0d-4b5f-add3-e12384d9dfe6/ (::1)
HTTPServerRequest(protocol='http', host='localhost:9000', method='DELETE', uri='/api/datasets/3ac211e9-4d0d-4b5f-add3-e12384d9dfe6/', version='HTTP/1.1', remote_ip='::1')
Traceback (most recent call last):
  File "/home/clausen/.virtualenvs/lt_qa/lib/python3.7/site-packages/tornado/web.py", line 1699, in _execute
    result = await result
  File "/home/clausen/.virtualenvs/lt_qa/lib/python3.7/site-packages/libertem/web/dataset.py", line 23, in delete
    await self.data.remove_dataset(uuid)
  File "/home/clausen/.virtualenvs/lt_qa/lib/python3.7/site-packages/libertem/web/base.py", line 144, in remove_dataset
    for job in self.jobs.values()
  File "/home/clausen/.virtualenvs/lt_qa/lib/python3.7/site-packages/libertem/web/base.py", line 145, in <listcomp>
    if self.dataset_to_id[self.dataset_for_job[job]] == uuid
KeyError: <SERDataSet for /home/clausen/iffcloud/Open Pixelated STEM framework/Data/3rd-Party Datasets/LinTian/1.ser>
```

Not sure how to reproduce, so I think low priority.

uellue, 2020-04-07:
@sk1p  Attribute to cosmic rays and close 😉, or postpone to 0.6?

sk1p, 2020-04-08:
Hmm, as the code that could have caused this has changed quite a bit, I'm going to close this. If this is still a problem, I guess we will see the error again.

closed 2020-04-08 by sk1p