LiberTEM/LiberTEM #602: Spurious performance regression for disk analysis with remote Dask cluster

uellue, 2020-02-14:
When using a remote Dask cluster running a disk analysis leads to a massive spurious performance regression in comparison to a plain mask analysis:

[image]

The same effect is observable in the GUI, both with `libertem-server` running on moellenstedt and on the Windows machine. 

Connecting to the same cluster from a Jupyter notebook on moellenstedt in the same Python environment triggers the error as well, meaning this is not a cross-platform or version mismatch issue.

An excessive size of the pickled tasks can be excluded judging from the output of debugging code. The regression occurs on repeated execution. The CPU is busy during that time, judging from the output of `top`.

It does not occur when running the same test on the same machine with a local cluster directly with JupyterHub. The performance with a local cluster is 4x (mask analysis) to 50x (disk analysis) better compared to the remote cluster over TCP to localhost.

[image]

It also doesn't occur when starting libertem-server with a local cluster on moellenstedt.

Versions: LiberTEM (nearly) master, dask and distributed 2.10.1 /2.10.0 (latest)

uellue, 2020-03-11:
- [ ] Test with stand-alone Dask, minimal reproducer
- [ ] Bisect

uellue, 2020-03-17:
The regression was caused by not setting the thread count for the stand-alone workers correctly. Refs #546

closed 2020-03-17 by uellue

sk1p, 2020-03-17:
Ugh, okay. Well done tracking this down, thanks!

uellue, 2020-03-17:
@sk1p Well, I dug out a new problem in the process: #684 🙄