LiberTEM/LiberTEM #609: Better error message for empty ROI

sk1p, 2020-02-18:
To reproduce, run a UDF with `roi=np.zeros(ds.shape.nav, type=bool)`. Currently raises a weird exception in the internals...

uellue, 2020-02-18:
Discussion with @sk1p :

We should probably emulate NumPy's behavior with arrays that have a zero in their shape. That means we create empty result buffers of the appropriate shape if the ROI is empty. Editing title accordingly.

Example code for NumPy empty arrays:

```python
In [1]: import numpy as np

In [2]: a = np.ones((0, 1, 0))

In [3]: a.shape
Out[3]: (0, 1, 0)

In [4]: a
Out[4]: array([], shape=(0, 1, 0), dtype=float64)
```

AnandBaburajan, 2020-04-25:
> To reproduce, run a UDF with `roi=np.zeros(ds.shape.nav, type=bool)`. Currently raises a weird exception in the internals...

I've managed to reproduce this error:
`ValueError: invalid extra_shape (0, 16, 16): cannot contain zeros`

That's the error right? I'm familiarising myself with UDFs and I've got @uellue's comment. 

@sk1p I'd like to work on this. Anything else I should know or keep in mind except uellue's comment? Thanks!

sk1p, 2020-04-26:
> That's the error right?

Sound right, yes.

> Anything else I should know or keep in mind except uellue's comment?

Don't think so - let me know if you have further questions when you start your implementation. Btw. a good approach for changes like these is to practice TDD and start by writing tests that initially fail, and then work to make the tests succeed.

PR #765 (2020-05-04): Handle empty ROI and extra_shape with zero
Fixes #609 

## Contributor Checklist:

* [x] I have added or updated my entry in [the creators.json file](https://github.com/LiberTEM/LiberTEM/blob/master/packaging/creators.json)
* [x] I have added [a changelog entry](https://github.com/LiberTEM/LiberTEM/tree/master/docs/source/changelog) for my contribution
* [x] I have added/updated test cases
closed 2020-05-04 by sk1p