LiberTEM/LiberTEM #623: GitHub pages: add organization page

sk1p, 2020-02-19:
https://libertem.github.io/ should be some kind of landing page that links to the documentation of our packages - so currently LiberTEM itself and the blobfinder. AFAIU this can be done by having a repository `libertem.github.io` in our organization. How to structure this and what tools to use?

See also: https://help.github.com/en/github/working-with-github-pages/about-github-pages#types-of-github-pages-sites

sayandip18, 2020-03-08:
Currently, there is a github.io page for Blobfinder. So, is our goal is to create a new website that has both LiberTem and the blobfinder? If so, we can use Jekyll.

uellue, 2020-03-09:
Since our documentation is built with sphinx so far, I'd prefer to stick with it for the organization page as well. That way we can use the familiar structure, syntax, and things like intersphinx links.

A quick research on what to do:

- [ ] Create empty `libertem.github.io` repo in the LiberTEM organization
- [ ] Create separate document tree in LiberTEM main repo or a dedicated repository for the sources
  * Separate repo: Only build when page is updated, not on every CI run
  * Main repo: Use existing Travis and Tox set-up
- [ ] Include additional build commands in `tox.ini` and instruct `travis-sphinx` to use this tree via `travis-sphinx build --source=other/dir/doc` and the target repo via `GH_REPO_SLUG = 'LiberTEM/libertem.github.io'`.

uellue, 2020-03-09:
> So, is our goal is to create a new website that has both LiberTem and the blobfinder?

An overview page for the project that links to both.

sk1p, 2020-03-09:
> Since our documentation is built with sphinx so far, I'd prefer to stick with it for the organization page as well

Agreed.

> Create empty libertem.github.io repo in the LiberTEM organization [...]
> main repo or a dedicated repository for the sources

I rather like the variant of using branches for this: have the documentation source and CI pipeline in the master branch of the `libertem.github.io` repository and directly visible in github, and put the result into the `gh-pages` repo. That way, the source is directly "attached" to the built pages and easily findable.

So that would mean copying a bit of infrastructure (sphinx setup, maybe but not necessarily tox, travis config) to the new repo, but for this purpose, it should be quite minimal: we likely only need to change it when we add a new repo/python package to the libertem organization. 

> Include additional build commands in tox.ini and instruct travis-sphinx to use this tree via travis-sphinx build --source=other/dir/doc and the target repo via GH_REPO_SLUG = 'LiberTEM/libertem.github.io'

This last point is then not necessary, the config can be a simplified version of what is in the main repo.

I went ahead and created the repository - https://github.com/LiberTEM/libertem.github.io - feel free to open a PR there!

We should maybe also discuss what each item on that landing page should consist of. From my side this should be at least:

 - title
 - short description what it is about
 - links: github repo, documentation, pypi

uellue, 2020-03-12:
https://help.github.com/en/github/working-with-github-pages/about-github-pages#publishing-sources-for-github-pages-sites

_"The default publishing source for user and organization sites is the master branch. If the repository for your user or organization site has a master branch, your site will publish automatically from that branch. You cannot choose a different publishing source for user or organization sites."_

Maybe we can reverse, with source in a side branch and published version in master?

uellue, 2020-03-12:
Hm, the target branch `gh_pages` seems to be hard-coded in `travis-sphinx`: https://github.com/Syntaf/travis-sphinx

Since updates will be infrequent, we can perhaps just pull from `gh_pages` into master manually?

sk1p, 2020-03-12:
We can also not use travis-sphinx - if we want to move to azure, we will need to use a different flow anyways

uellue, 2020-03-12:
Good point, maybe a good time to test this at this opportunity?

sk1p, 2020-03-12:
Indeed! Mostly this involves finding out how the authentication will work from both travis and azure, as we will need to push the results to a git branch. I guess the `GH_TOKEN` that is available in the CI should be enough?

uellue, 2020-04-09:
Depends on #565

uellue, 2021-09-06:
The page exists now with quite a bit of content. Certainly can still be improved, but we can maybe close here at this point.

closed 2021-09-06 by uellue