LiberTEM/LiberTEM #627: Hackathon "postprocessing"

sk1p, 2020-02-20:
- [ ] update documentation from hackmd document
- [ ] update credits

sayandip18, 2020-03-06:
Isn't this issue same as #638 ?

uellue, 2020-03-07:
> Isn't this issue same as #638 ?

#638 is a specification and spin-off from this issue since @sk1p and I realized that this would be a bigger item.

sayandip18, 2020-03-08:
>     * [ ]  update credits
 
What does this mean?

sk1p, 2020-04-10:
> What is meant by updating credits?

Sorry we didn't answer this before. This was done in https://github.com/LiberTEM/LiberTEM/pull/708 - basically it means updating our contributor and creator files and crediting people from the hackathon for their contribution.

uellue, 2021-05-27:
Closing this overview issue since the remaining issues have their own entry and can be adressed separately.

closed 2021-05-27 by uellue