LiberTEM/LiberTEM #628: Allow final postprocessing on the central node in an UDF

uellue, 2020-02-20:
Discussion with @sk1p and @Brow71189:

Currently, final postprocessing of an UDF's results is performed in a `run_*()` function that is not part of the UDF. If live updating results are required, this has to be re-implemented by the routine that is consuming the live-updating raw results.

In the future, an UDF should be able to specify a final processing routine that transforms raw results into something useful for the user. `StdDevUDF` would be a prime non-trivial example where this is required.

It should be possible to perform the transformation for each merging step so that incremental postprocessed results are available.

sk1p, 2020-03-24:
Just for naming: how about `UDF.finalize()`?

One important point: making this a method on the UDF will currently limit us to finalizing only using the result of one UDF. This is not necessarily bad, just something to keep in mind.

uellue, 2020-03-25:
`finalize()` is the name of a destructor in many languages: https://en.wikipedia.org/wiki/Finalizer.

What about `consolidate()`?

uellue, 2020-03-25:
It should probably be a pure function that doesn't modify the state of the UDF. In that case we can call it whenever we need consolidated partial results.

sk1p, 2020-03-25:
> It should probably be a pure function that doesn't modify the state of the UDF. In that case we can call it whenever we need consolidated partial results.

:+1:

> finalize() is the name of a destructor in some languages

Tried my best to forget about Java :grin:

> What about consolidate()?

Hmm, to me this sounds like a merge/reduce-ish operation, but really this could be any kind of transformation. What about the more general `get_results`? It may also make it clear that it is a getter and not really a function where you should put anything computationally intensive

uellue, 2020-04-09:
> What about the more general `get_results`?

:+1:

PR #994 (2021-03-18): Support for a final postprocessing on main node
Add `UDF.get_results` method that is called on the master node after each merge. Fixes #628

Note that for the `StdDevUDF`, the implementation is completely backwards-compatible - this is a bit awkward, as the arrays are returned either as `ndarray` or as `BufferWrapper` . Other UDFs may have similarly decisions to make. Especially with `roi`s this may be a bit problematic - the UDF has to decide if it wants to return the derived result as `raw_result`-like array, or embedded into a `NaN`-filled array.

For other UDFs, where backwards-compatibili [... cut]
commit d01dc859 (2021-03-18): Support for a final main node postprocessing

Add `UDF.get_results` method that is called on the master node after
each merge. Fixes #628
closed 2021-03-18 by uellue