LiberTEM/LiberTEM #629: Releasing docs: add item about handling deprecations

sk1p, 2020-02-20:


twentyse7en, 2020-02-23:
I would like to contribute.

sk1p, 2020-02-23:
> I would like to contribute.

Okay! Did you already find the release checklist? This issue is about adding new items in there to remind us to take care of deprecations.

sk1p, 2020-02-24:
Another point for the release checklist: Add the current LiberTEM version to CVL @ https://github.com/Characterisation-Virtual-Laboratory/CharacterisationVL-Software

twentyse7en, 2020-02-24:
Where can I find release checklist ?

sk1p, 2020-02-24:
It [on the releasing page](https://libertem.github.io/LiberTEM/releasing.html#release-checklist), which is linked from [Contributing -> Advanced](https://libertem.github.io/LiberTEM/contributing.html#advanced)

twentyse7en, 2020-02-25:
So I need to add items in checklist which will removed in release 0.5, right ?

sk1p, 2020-02-25:
The idea is to add general checklist items for handling deprecation and publishing to CVL, not specifically for one version

twentyse7en, 2020-02-25:
For finding `DeprecationWarning` I have gone through `LiberTEM/src`. Do I have to search in entire code(`LiberTEM/`) or just `Libertem/src`

sk1p, 2020-02-25:
Hmm, you are overthinking it. The idea is just to have a *very* simple item in the release checklist to remind us to take care of deprecations. There doesn't need to be a list of concrete warnings at all, and also no need for a detailed description: it is supposed to be terse. The target audience of the release checklist are experienced contributors of LiberTEM who want to either a) do a release themselves or b) help out with doing the quality assurance.

twentyse7en, 2020-02-26:
Thx, I was going in a different direction :sweat_smile: Hope the changes made in PR are something you expects.

closed 2020-03-02 by sk1p