LiberTEM/LiberTEM #630: Make `Shape` and `Slice` a bit more userfriendly

sk1p, 2020-02-20:
- [ ] `UDF.buffer` should accept `Shape` as argument `extra_shape`
- [ ] allow concatenation of `Shape` and `tuple` (what rules should we define? should we extend the nav or sig dims? or should we return a plain `tuple`?)

adlakha-rachit, 2020-03-02:
Hello,
I would like to work on this.
Can you please help me get started as I am new to LiberTEM?

sk1p, 2020-03-02:
Hi @adlakha-rachit - welcome to LiberTEM, I'd love to help you get started! As a first step, it is a good idea to [install LiberTEM](https://libertem.github.io/LiberTEM/install.html). We also have a documentation section [dedicated on contributing to LiberTEM](https://libertem.github.io/LiberTEM/contributing.html), which every contributor should have read at least once.

Let us know if anything is unclear in the documentation!

If you have trouble getting started with this specific issue, it is a good idea to try to familiarize yourself with the mentioned concepts and Python objects (`UDF`, `UDF.buffer`, `Shape`, ...). This may require reading more of our documentation, or having a look at the specific source files. Try to understand the exact meaning of the issue description, and why the described change is necessary. 

If you have more specific questions, please let us know! Also, feel free to [join our gitter channel](https://gitter.im/LiberTEM/Lobby), to get to know the LiberTEM developers and other students. The gitter chat is also a good place to ask more general questions, which are not related to a specific issue.

saisunku, 2020-03-24:
Hi @sk1p and @uellue, as per the documentation here https://libertem.github.io/LiberTEM/udf.html `UDF.buffer` already accepts `extra_shape` as a parameter. Is the first part of this issue resolved then?

For the second part, what `tuple` are you referring to?

sk1p, 2020-03-24:
> UDF.buffer already accepts extra_shape as a parameter

It does, but: in LiberTEM we have a few helper classes, one of which is `Shape` that represents an array shape but additionally specifies the number of "signal dimensions". We try to explain this on [our concepts page](https://libertem.github.io/LiberTEM/concepts.html) (let us know if this is unclear!

The API of `Shape` is explained here: https://libertem.github.io/LiberTEM/reference/internals.html#shapes-and-slices

Now, if you try to pass a `Shape` object as `extra_shape`, you should get some unfriendly exception - that is what the first part is about, the internals should be modified to also accept `Shape` objects.

> For the second part, what tuple are you referring to?

This is about the general operation: addition of two `tuple`s again results in a tuple, via concatenation:

```
>>> (1, 2, 3) + (4, 5, 6)
(1, 2, 3, 4, 5, 6)
```

Now if you try to do this with a `tuple` and the aforementioned `Shape` object, you will most likely also get some unfriendly error message, like:

```
TypeError: unsupported operand type(s) for +: 'Shape' and 'tuple'
```

So the second part of the issue is to define and implement some sane concatenation operations between tuples of integers and `Shape` objects, hopefully making our APIs that return `Shape` objects more user friendly.

saisunku, 2020-03-24:
I see! I understand navigation and signal dimensions. As far as I understand, `extra_shape` in buffer is the shape of the result of the calculation that the UDF is doing. So there are no navigation and signal dimensions. So then how do we accept a `Shape` object to the buffer? Should we check that this incoming `Shape` object has `sig_dims = 0`?

sk1p, 2020-03-24:
> I see! I understand navigation and signal dimensions.

:+1:

As far as I remember, the use case was slightly hacky; the buffer definition looked a bit like this:

```
def get_result_buffers(self):
  return {
    "buf": self.buffer(kind="single", dtype=int, extra_shape=self.meta.dataset_shape.nav)
  }
```

So in that case it did indeed have `sig_dims=0`.

> extra_shape in buffer is the shape of the result of the calculation that the UDF is doing. So there are no navigation and signal dimensions

To clarify, `extra_shape` is the *additional* shape, on top of the "base shape" defined by `kind="..."`.

> Should we check that this incoming Shape object has sig_dims = 0?

`kind="single"` is often used as an "escape hatch" for things that don't (yet) fit directly into the UDF programming model - so possibly `sig_dims` could be non-zero, for example if you are for some reason passing in `dataset_shape.sig` with `kind="single"` instead of `kind="sig"`... so I would say just "blindly" converting to tuple and discarding the `sig_dims` information could be justified.

saisunku, 2020-03-26:
Thanks @sk1p. Regarding the concatenation of `Shape` and `tuple`, perhaps we could extend the navigation dimensions and leave the signal dimensions as is. One possible use I can think for this is to combine multiple datasets (e.g. different materials or different parameters) taken on the same detector. In that case, you would extend the navigation dimensions.

saisunku, 2020-03-26:
I make PR (#694) for the first part. Let me know what you think!

sk1p, 2020-03-28:
> Regarding the concatenation of `Shape` and `tuple`, perhaps we could extend the navigation dimensions and leave the signal dimensions as is? One possible use I can think for this is to combine [...]

Hmm, I'm not sure. Perhaps we should leave this open for now and revisit this once we have concrete use cases. Alternatively, one could do a survey in the LiberTEM sources themselves and look for `Shape` + `tuple` concatenations - I think there should be a few.

saisunku, 2020-03-28:
@sk1p thanks so much for the detailed comments on the PR #694 . I will update it.

> Hmm, I'm not sure. Perhaps we should leave this open for now and revisit this once we have concrete use cases. Alternatively, one could do a survey in the LiberTEM sources themselves and look for [...]

Could you briefly tell me how to survey the LiberTEM source for `Shape` + `tuple` concatenations?

sk1p, 2020-03-28:
Usually, this would be done with static code analysis tools, but for smaller code bases, simple searching with `grep` or similar tools using a regular expression should also work.

Existing `Shape + tuple` should all involve converting the `Shape` to a `tuple`, so something like this should give you some potential instances of this conversion:

`shape.*tuple|tuple.*shape`

Then have a look at the code to try and "infer" the types involved, possibly also looking at docstrings of functions, or even their call sites, if they are not clear. Opposed to a proper static analysis, this won't find some instances, like multi-line statements, or those with differently named variables...

uellue, 2020-04-09:
@sk1p Where would the concatenation be used? Do you have a specific example where the lack of this feature makes things complicated?

sk1p, 2020-04-09:
It's not complicated, you just have to get used to it. In the last few days, I typed `some_other_tuple + tuple(some_Shape_instance)` quite a few times, but someone new to LiberTEM might stumble over that.

uellue, 2020-04-09:
Ok I see :-)

By the way, for anyone interested in this: https://www.python-course.eu/python3_magic_methods.php

saisunku, 2020-04-09:
@sk1p @uellue sorry for the dropping the ball on this, I got busy with research. I will work on it this weekend. I will first find instances of concatenation in existing code and compile a summary of them.

saisunku, 2020-04-13:
@sk1p I used your regular expression `shape.*tuple|tuple.*shape` to search the codebase and compiled a list of `tuple` and `Shape` object concatenations. Most of them are for `Shape.sig` but there are a few that are `Shape.nav` or the entire `Shape` object. As you said, this method doesn't catch all of the concatenations. Let me know what you think. Thanks.

`Shape.sig + tuple`
prototypes\smallfiles\raw_parts.py line 84
prototypes\smallfiles\raw_parts.py line 138
src\libertem\api.py line 511
src\libertem\analysis\raw.py line 86
src\libertem\common\slice.py line 273
src\libertem\common\slice.py line 293
src\libertem\io\dataset\blo.py line 175
src\libertem\analysis\clust.py line 106
src\libertem\io\dataset\cluster.py line 162
src\libertem\io\dataset\hdf5.py line 230
src\libertem\io\dataset\hdf5.py line 261
src\libertem\io\dataset\hdf5.py line 292
src\libertem\io\dataset\hdf5.py line 320
src\libertem\io\dataset\base\part3d.py line 226
src\libertem\io\dataset\base\part3d.py line 292
src\libertem\io\dataset\base\part3d.py line 296
src\libertem\io\dataset\base\part3d.py line 305
src\libertem\io\dataset\base\part3d.py line 355
src\libertem\io\dataset\base\part3d.py line 370
src\libertem\io\dataset\base\part3d.py line 392
src\libertem\io\dataset\base\partition.py line 84

`Shape.nav + tuple`
src\libertem\analysis\masks.py line 182
src\libertem\job\masks.py line 164
src\libertem\job\masks.py line 168

`Shape.flatten_nav().nav + tuple`
src\libertem\job\masks.py line 78

Unclear
src\libertem\io\dataset\dm.py line 232 - looks like whole shape obj?
src\libertem\io\dataset\ser.py line 121 - looks like whole shape obj?
src\libertem\udf\base.py line 676 - shape[1:]

sk1p, 2020-04-14:
@saisunku Okay! Looks like the `Shape.sig + tuple` case is the most common.

Coming back to the original question:

> Regarding the concatenation of Shape and tuple, perhaps we could extend the navigation dimensions and leave the signal dimensions as is? One possible use I can think for this is to combine multiple [...]

Hmm, I would prefer if `__add__` blindly converted to a tuple, and if we kept the behavior of tuple addition, so the Shape.sig + tuple cases above can be written as ``(1, 2, 3) + shape.sig`` in the future. While this is the most common case, and _could_ be implemented as ``(1, 2, 3) + shape`` instead, I would prefer not to do this, as it is not clear that it would replace the existing navigation dimensions.

Extending the navigation dimensions for combining data sets, as described, would be a completely different operation (stacking?) - it would possibly introduce a new dimension, or adding to an existing one. Like this:

```python
>>> s1 = Shape((32, 32, 512, 512), sig_dims=2)
>>> s2 = Shape((32, 32, 512, 512), sig_dims=2)
>>> s3 = Shape((32, 32, 512, 512), sig_dims=2)
>>> stacked_1 = stack(s1, s2)  # mock API, not really how I would do it...
>>> stacked_1
Shape((2, 32, 32, 512, 512), sig_dims=2)
>>> stacked_2 = stack(stacked_1, s3)
>>> stacked_2
Shape((3, 32, 32, 512, 512), sig_dims=2)
>>> # etc.
```

This hasn't come up yet, as our UDF built-in functionality currently mostly works on single data sets, but may be something to keep in mind for the future. I would not want to do this in the `__add__` operator, though.

What do you think?

saisunku, 2020-04-14:
@sk1p 

> Hmm, I would prefer if `__add__` blindly converted to a tuple, and if we kept the behavior of tuple addition, so the Shape.sig + tuple cases above can be written as `(1, 2, 3) + shape.sig` in the [...]

This makes sense. One more idea. How about implementing `__add__` only when either `sig_dims` or `nav_dims` are zero? If both of these are not satisfied, we return `NotImplemented`. This will ensure that there is no confusion but `tuple + Shape.sig` and `tuple + Shape.nav` will still work as expected.

sk1p, 2020-04-14:
> How about implementing `__add__` only when either sig_dims or nav_dims are zero?

:+1: :+1: - other cases are then still possible by explicitly "casting" the `sig_dims` away by calling `tuple` on the `Shape` - perfect!

> NotImplemented

Dynamically raising `NotImplemented` I would disagree with - it can't be fixed by implementing a method somewhere. How about `ValueError` instead? From [the description](https://docs.python.org/3/library/exceptions.html): "Raised when an operation or function receives an argument that has the right type but an inappropriate value, and the situation is not described by a more precise exception such as IndexError"

saisunku, 2020-04-14:
OK! I will implement this and make a pull request.

saisunku, 2020-04-20:
@sk1p Another question. What about adding two `Shape` objects? Could we implement that as a concatenation of both the navigation and signal dimensions?

sk1p, 2020-04-20:
Sounds sensible, if we still  keep the constraint of having nav or sig 0. This would be to support something like `s3 = s1.nav + s2.sig`, right?

saisunku, 2020-04-20:
@sk1p I was thinking `S = S1 + S2` where `S.sig = S1.sig + S2.sig` and `S.nav = S1.nav + S2.nav`

sk1p, 2020-04-23:
> I was thinking `S = S1 + S2` where `S.sig = S1.sig + S2.sig` and `S.nav = S1.nav + S2.nav`

The question here would be what the concrete use case is. I think it would be better to postpone this until we do have a use case, and for now first implement the other cases.

saisunku, 2020-04-23:
@sk1p OK, thanks! Please see PR #749

sk1p, 2021-03-24:
I'm actually quite happy with this now, closing.

closed 2021-03-24 by sk1p