LiberTEM/LiberTEM #631: UDF: attribute access in merge

sk1p, 2020-02-20:
Instead of requiring `dict`-like access, also allow access by attribute name, i.e. `dest.pixelsum_nav_raw[:] += src.pixelsum_nav_raw`

Otherwise there is a strange disconnect between the `process_*` functions and `merge`.

sayandip18, 2020-04-30:
Seems like `sum.py`, `logsum.py`, `raw.py` and `masks.py` are the only files that have this kind of access.  
The rest should be modified in a way to allow this, right?

sayandip18, 2020-05-01:
What exactly are the purpose of these quotation marks inside the function [here](https://github.com/LiberTEM/LiberTEM/blob/bd29333c745b39186d0bc04bc0c9e0d701752fa8/src/libertem/udf/sum.py#L31)?

sk1p, 2020-05-04:
> What exactly are the purpose of these quotation marks inside the function [here](https://github.com/LiberTEM/LiberTEM/blob/bd29333c745b39186d0bc04bc0c9e0d701752fa8/src/libertem/udf/sum.py#L31)?

They are the docstring of the method, so in this case, they override the docstring of the base class to be an empty string. We do this sometimes if it doesn't make sense to have the docstring of the base class visible in the documentation of the subclass.

sk1p, 2020-05-04:
> Seems like `sum.py`, `logsum.py`, `raw.py` and `masks.py` are the only files that have this kind of access.
> The rest should be modified in a way to allow this, right?

I'm not sure I follow - none of the `merge` functions can use the attribute-way of accessing the buffers right now, they all require `dict`-like access.

sayandip18, 2020-09-08:
> 
> I'm not sure I follow - none of the `merge` functions can use the attribute-way of accessing the buffers right now, they all require `dict`-like access.

My bad. I am still a bit confused though. The `process_*` methods in the utility UDF classes have `self.results.buffername` (like [here](https://github.com/LiberTEM/LiberTEM/blob/d19b4e5d6cf1ebae929575381debc4b59bdf342c/src/libertem/udf/sumsigudf.py#L28)). I get it that it is accessing the dictionary returned by `get_result_buffers()`, but there is no attribute called `results` in UDF class!

PR #1003 (2021-03-25): UDF: Consistency fixes and `get_results` improvements
Now, all accesses to buffers in `UDF.results` and the `src`/`dest` parameters of `UDF.merge` should be done via `__getattr__`; if the `BufferWrapper` is really needed, then `UDF.results.get_buffer` should be used.

As a follow-up to #994, `get_results` now should consistently return numpy arrays, which are then converted to `BufferWrapper`s before returning them to the user. This means we can make `get_results` a bit more DRY by not having to repeat buffer names. Also, buffers that don't have to be post-processed, can be left out in `get_re [... cut]
closed 2021-03-25 by uellue