LiberTEM/LiberTEM #633: Annular CoM detection in GUI

matkraj, 2020-02-21:
Hello @uellue and @sk1p,
It would be beneficial for some applications (Lorentz STEM) to include annular mask for centre of mass imaging. This would allow filtering of large proportion of diffraction contrast from magnetic and electric field imaging in nano-diffraction. It would be a great feature for live imaging applications. The technique was first introduced with annular DPC detectors in https://doi.org/10.1109/20.104427.

uellue, 2020-02-21:
@matkraj That's a nice point! With the current version one can do this with scripting. Implementing this in the GUI is not too difficult, too. I'm including this in our GSoC ideas list, together with the other remaining items of #31 .

Is this more of a general idea or would this be used asap by someone?

How does this compare with correlation-based detection of the shift?

matkraj, 2020-02-21:
@uellue It is a more general idea. It will be a very useful feature when live imaging. 

Gradient cross-correlation will be always better. Interestingly, the two techniques can be somehow similar if one could make an annular detector with variable position and beam only overlapping a tiny proportion over the inner quadrants.

uellue, 2020-02-21:
> It will be a very useful feature when live imaging.

Actually, the correlation will also work with live data in LiberTEM. Just missing a GUI at the moment. 😄

PR #1089 (2021-08-25): Annular CoM
Fixes #633

## TODO

- [x] update screenshot in `app/com`

## Contributor Checklist:

* [x] I have added or updated my entry in [the creators.json file](https://github.com/LiberTEM/LiberTEM/blob/master/packaging/creators.json)
* [x] I have added [a changelog entry](https://github.com/LiberTEM/LiberTEM/tree/master/docs/source/changelog) for my contribution
* [x] I have added/updated documentation for all user-facing changes
* [x] I have added/updated test cases
* [x] I have included the [rebuilt production build of the client](https://libertem.github.io/LiberTEM/contributing.html?#b [... cut]
commit c7be4615 (2021-08-25): Support for annular CoM, fixes #633

It is supported both in the GUI and via the `Context` API.
closed 2021-08-25 by sk1p