LiberTEM/LiberTEM #635: Information on tiling scheme in UDF

uellue, 2020-02-21:
Discussion with @sk1p :

The tiled `StdDevUDF` #625 requires a per-tile frame counter. In the current version, this is solved with a somewhat awkward `dtype='object'` buffer and custom dynamic data structure. In the future, information on the tiling scheme should be available as meta information. That allows to allocate the buffers in `get_result_buffers()` or `get_task_data()`.

Ideas by @sk1p :

[image]

[image]

uellue, 2020-04-14:
Discussion with @sk1p: The `StdDevUDF` currently expects a constant tiling scheme in signal direction across the entire dataset. With negotiation and GPU support #364, hlding this assumption will no longer be practical or desirable.

Proposed solution:

* A constant tiling scheme in signal direction is guaranteed only within the partition.
* The per-tile frame counters in `StdDevUDF` are implemented as task data buffers.
* Frame counters per partition for the final merging step are kept separately as `kind=single` without tiling
* The task data per-tile counters are merged into the partition buffer in `postprocess()` where it is guaranteed that all tiles of the partition have been processed. In this particular case, it means that all per-tile counters should be equal. For the final merging on the central node, the per-tile information is therefore irrelevant.

sk1p, 2020-04-14:
How does the tile index and slice end up in the UDF?

Instead of passing only `tile.data` into `process_tile`, pass the `DataTile` instance. `DataTile` then needs to know the index in the scheme, but can also have a reference to the signal slice. Instead of putting the data into a `.data` attribute, `DataTile` should inherit from `np.ndarray` and behave like any other numpy array.

Later we can also add information about the navigation position in a useful way.

sk1p, 2020-04-16:
I updated the `Stddev` UDF for the new `TilingScheme` API in 0ed27423986b866e73c15a4c230e1232281a4d30, please have a look and let me know if this is correct.

PR #706 (2020-04-27): WIP: Tiled reading in more file formats
Fixes #554, fixes #634, fixes  #27, fixes #331, fixes #373, fixes #635, fixes #435, fixes #247 

## Functionality implementation

- [x] Update `MemoryDataSet` #634 
- [x] Create API for tileshape negotiation #554 
- [x] Add tiling scheme and current index to `UDFMeta` #635 
- [x] Update all `DataSet` impls to work with the new infrastructure
  - [x] hdf5 - can't use the I/O backend as I/O is done by HDF5 itself
  - [x] ser - I/O is done by ncempy, so we can't use the I/O backend
  - [x] raw
  - [x] mib
  - [x] blo
  - [x] k2is
  - [x] frms6
  - [x] empad [... cut]
closed 2020-04-27 by sk1p