LiberTEM/LiberTEM #637: Add documentation for get_diagnostics()

uellue, 2020-02-21:
A dataset's `get_diagnostics()` method can help to find additional datasets in a HDF5 file.

Side note -- is the info button in the GUI documented that provides the same info?

sk1p, 2020-02-21:
The GUI [uses](https://github.com/LiberTEM/LiberTEM/blob/05edd76e46ef7dbeafa50fa9332809c75767dcdc/src/libertem/web/base.py#L183) [the diagnostics property](https://github.com/LiberTEM/LiberTEM/blob/05edd76e46ef7dbeafa50fa9332809c75767dcdc/src/libertem/io/dataset/base.py#L240) - which also contains some general information about partitioning. I guess that's also the one you would access as an API user. Note that the information therein is not meant for automated consumption in scripts (note also the weird list-of-dict structure) and is not guaranteed to be stable!

sayandip18, 2020-04-05:
Found a typo in [line 242](https://github.com/LiberTEM/LiberTEM/blob/05edd76e46ef7dbeafa50fa9332809c75767dcdc/src/libertem/io/dataset/base.py#L242)

sk1p, 2020-04-05:
> Found a typo in [line 242](https://github.com/LiberTEM/LiberTEM/blob/05edd76e46ef7dbeafa50fa9332809c75767dcdc/src/libertem/io/dataset/base.py#L242)

Nice catch, feel free to open a PR!

sayandip18, 2020-04-05:
This is strange! There's no file named base.py in my repository. Do I have to create the entire file?

sk1p, 2020-04-06:
> This is strange! There's no file named base.py in my repository

That's because the `base.py` file has been broken up [into more but shorter files](https://github.com/LiberTEM/LiberTEM/tree/master/src/libertem/io/dataset/base)

sayandip18, 2020-04-06:
> 
> If you update to the current master, you too should have a `base.py` file.
 
The current master doesn't have the `base.py` file. Also, the [link](https://github.com/LiberTEM/LiberTEM/blob/05edd76e46ef7dbeafa50fa9332809c75767dcdc/src/libertem/io/dataset/base.py) points to a "tree", I think.

sk1p, 2020-04-06:
> The current master doesn't have the base.py file. 

Ah, sorry. This:

> If you update to the current master, you too should have a base.py file.

was wrong, because the current version _no longer_ has the `base.py` file. But you can find the same classes in that directory of smaller files (you can have a look [at the `__init__.py` file](https://github.com/LiberTEM/LiberTEM/blob/master/src/libertem/io/dataset/base/__init__.py) as a kind of "index" of the package).

> Also, the link points to a "tree", I think

A tree is how git maps directories to its internal data structures. So any link to a directory on github will be a link to a tree. In this case, [the link](https://github.com/LiberTEM/LiberTEM/blob/05edd76e46ef7dbeafa50fa9332809c75767dcdc/src/libertem/io/dataset/base.py) is actually not a tree, but a "blob", which is how git stores specific versions of files (in this case, version `05edd`...).

If you are interested how trees, blobs etc. in git work, I can recommend [the official documentation on git internal objects](https://git-scm.com/book/en/v2/Git-Internals-Git-Objects).

> What about I just copypaste the entire thing and replace the smaller files with a single base.py file? Wouldn't that be the simplest way to get it done?

While it may be simple, it will undo my work of actually separating them. We try to keep files small in order to make classes and functions easier findable. It is hard to navigate files that are longer than a few hundred lines.