LiberTEM/LiberTEM #638: Restructure documentation along user's needs

uellue, 2020-02-21:
The HackMD notebook of the Swift hackathon is a nice, brief run-down on getting started with LiberTEM. Much of the information is available in our documentation, in principle. However, we can improve on providing a coherent flow and making the relevant info findable where a user would expect it.

* Tutorial-style quick introduction like the HackMD notebook
* Better links to examples?
* Conceptual reference documentation with a complete overview
* API reference documentation.

Refs #297

sayandip18, 2020-03-04:
I would love to contribute. Where do I start? So far, I've created a workspace in HackMD although am not sure what "HackMD notebook" means.

sk1p, 2020-03-04:
The hackmd notebook referenced is [this document](https://hackmd.io/3jO3rEm4TfaPhrJm4TeZcQ). It contains notes about our recent hackathon. The idea is to use the contents of the hackmd document as a guide to improve our official documentation. Please don't edit the hackmd notebook, as it should keep reflecting the state at the time of the hackathon.

sayandip18, 2020-03-06:
Will the changes be in README.rst?

sk1p, 2020-03-06:
> The official documentation is in github.io. Is it possible for me to change it?

Sure! The documentation sources are [part of this repository](https://github.com/LiberTEM/LiberTEM/tree/master/docs/source), so you can do them like any other pull request.

sk1p, 2021-01-27:
@uellue should we have another look at this for the 0.6 release? In #939 the docs already have a better structure, but I feel there is still quite a bit missing.

uellue, 2021-05-27:
See also #427