LiberTEM/LiberTEM #645: Proper error handling when opening certain files

sk1p, 2020-02-23:
If you try to open a dataset that is supported in LiberTEM itself, but not yet in the GUI (currently, that is  just DM), there is no proper error handling in place yet:

https://github.com/LiberTEM/LiberTEM/blob/b20b21dd455ae14d6f202259b995231e3f889e69/client/src/dataset/components/DatasetOpen.tsx#L159

THis is because for a long time, this truly could not happen. Instead, this should trigger a proper error message and dialog (like other errors).

sk1p, 2020-02-23:
To prevent duplicate work: @AnandBaburajan has expressed interest to work on this issue.

AnandBaburajan, 2020-03-01:
I was able to reproduce the issue and I see that an empty page is returned when a DM file is loaded.
I've two options in mind to solve this:

1. Show a pop-up box with a message similar to "DM files are currently not supported in the GUI".
2. Create a new DatasetType in the GUI for "DM" and a corresponding form page (DMParamsForm) with the message "DM files are not currently supported in the GUI", instead of a form.

Please let me know which one to go ahead with or if there's a better way to solve this. Thanks!

sk1p, 2020-03-01:
I would prefer option 1, using the existing error reporting facilities. If you haven't seen an error message in the GUI yet, an easy way to produce one is to enter an invalid path into the file browse  dialog and press return.

AnandBaburajan, 2020-03-04:
Sorry, It's been a few days with this issue. I've been trying to create a check for the "DM file detected on the GUI" error in [dm.py](https://github.com/LiberTEM/LiberTEM/blob/master/src/libertem/io/dataset/dm.py), but failed. I actually planned to check for the error in dm.py, then let redux store it and let the error show up in ErrorList. I'm not completely familiarized with how the Python API triggers the errors and interacts with Redux yet, but I'm learning.

Did I get the wrong idea? Can you briefly guide me with hints on how to approach this issue, whenever you're free? Thanks!

sk1p, 2020-03-04:
> Sorry, It's been a few days with this issue

Don't worry!

> I actually planned to check for the error in dm.py

I don't think this can work, as the dm `DataSet` doesn't know if it is used in a GUI or from another context (script, embedded into an application, notebook, ...). It could work to signal from the Web API whether a certain file format is supported in the GUI (for example via the `detect` call), but that would mean the Python code suddenly has to care about the GUI, and when support is added in the GUI, two components need to change, backend and frond-end. I would prefer if the GUI would decide this on its own.

> let redux store it and let the error show up in ErrorList

That part is :ok_hand: - you just need to trigger this from the GUI itself and not the Python code.

PR #666 (2020-03-10): Error handling for unsupported datasets in GUI
Fixes #645 

## Contributor Checklist:

* [x] I have added [a changelog entry](https://github.com/LiberTEM/LiberTEM/tree/master/docs/source/changelog) for my contribution
closed 2020-03-10 by sk1p