LiberTEM/LiberTEM #648: Bring LiberTEM to Conda

uellue, 2020-02-25:
We should allow installation via conda and via pip.

@reimarbauer, @sk1p  mentioned you would know how to approach this? :-)

ReimarBauer, 2020-02-25:
I guess we can get it on conda-forge, https://conda-forge.org/ without the pypi step first. If it can be builded on pypi too, the src path can be exchanged afterwards. 

We have to verify a few things first:
 - add conda-forge to your search path,
 - check by conda search if all in "install_requires" needed packages are available (check version), if one is missing we start with adding this to conda-forge.
 - libtertem should have some simple test, e.g. import something, or command -h so we can see that the CIs did everything well on building it
 - I guess we can use your implementation of setup.py. You may be interested for the js client to look at xstatic https://xstatic.readthedocs.io/en/latest/ May be you could create a seperate package for this. That would make it possible to install that package like any other setup.py based package. 

You can locally try out if your meta description works, thats how we did:
https://bitbucket.org/wxmetvis/mss/src/stable/localbuild/
see there "Testing local build" We improved the docs, for the next release ;)
https://bitbucket.org/wxmetvis/mss/src/stable/docs/development.rst
 
After your "alpha" performs as expected, do the fork of the staging recipes and add your recipe in a new branch and send a PR.

ReimarBauer, 2020-02-25:
For pypi, it is also quite easy, if all libraries for all platforms are there.

deepl.com may be used to translate to englisch ;) 
see at: Rezept Nr. 5: Ein Projekt auf PyPI registrieren
https://chrisarndt.de/talks/cooking-eggs/cooking-eggs-vortrag.html

sk1p, 2020-02-25:
> For pypi, it is also quite easy, if all libraries for all platforms are there.

PyPI is [already done](https://pypi.org/project/libertem/) :heavy_check_mark: - only conda is still missing. The main point would be proper integration of our CI pipeline and the whole conda build setup.

sk1p, 2020-03-04:
From offline discussion with @ReimarBauer, thank you!
 
 - we need to fork staged-recipe and create recipe for LiberTEM, which later will become a libertem-feedstock (see for example https://github.com/conda-forge/staged-recipes/pull/10807 or https://github.com/conda-forge/staged-recipes/pull/10961)
 - once we have a feedstock, a bot will take take of kicking off updates to it, anytime we push a new version to pypi

> check by conda search if all in "install_requires" needed packages are available (check version), if one is missing we start with adding this to conda-forge.

I've added this to a TODO list in the top issue description.

> libtertem should have some simple test, e.g. import something, or command -h so we can see that the CIs did everything well on building it

I guess we can use `libertem-server --help` for this, although it doesn't import a lot from libertem. Maybe additionally `python -c 'import libertem.api'`?

> You may be interested for the js client to look at xstatic

Hmm - I'm not sure I understand what we would gain by using xstatic. Right now, we are using about 15 packages from npm, which we build together with our code into a single bundle using webpack. This bundle is then shipped with our Python sdist and wheel. If I understand correctly, using xstatic, we would replace the official npm packages with xstatic packages from PyPI, which are maintained separately from the npm repository (either by us or a third-party). Please correct me if I have the wrong impression.

I see the benefit and like the idea, if you are using the "classical" way of front-end dev, that is, just put a few script tags into your HTML page, and manually manage versions and dependencies. Sadly, this doesn't work well with the way current front-end projects are built: using npm for dependencies, and using a build tool like webpack to bundle and optimize the whole program (for example, tree shaking to remove unused code, and sharing dependencies over the whole dependency tree). You also lose the timely and easy updates you get from npm - the packaged React version is 5 years out of date :frowning:, while with npm I can type `npm update` and get the react release from a week ago.

ReimarBauer, 2020-09-04:
Because I'm working currently on a simple web frontend for our MSS project I prepared the XStatic base and xstatic-bootstrap and xstatic-jquery for conda-forge. In our case listing the dependencies in a meta.yaml and install the latest version is at current state for our project similar to the existing dependencies of other python packages.

I understand your point. A solution is only to have more maintainers of those packages on pypi while on the other site for pure npm related packaging this is an amount of overhead not needed. 

Two pathes, many different use cases ;)

sk1p, 2022-06-20:
Closing in favor of current discussion in #1271