LiberTEM/LiberTEM #657: stddev stability issue

sk1p, 2020-02-29:
See the following CI failure: https://travis-ci.org/LiberTEM/LiberTEM/jobs/656632518

Not sure if this is fixed in #640.

sk1p, 2020-03-01:
See also: https://ci.appveyor.com/project/sk1p/libertem/builds/31162834/job/b5jxjmtkt13ii6l2

uellue, 2020-03-09:
Lovely! </irony> Those are not small errors anymore. We should definitely observe if this happens again with the new code. How can we make the error reproducible, i.e. get the same state of the random number generator etc.?

Or perhaps we could add code similar to https://github.com/LiberTEM/LiberTEM/blob/afe3c4292a2d55d31b0b2177aff0fa5e0cc9cb51/tests/test_analysis_masks.py#L932-L1011 to test standard deviation numerical stability with a test case that is sensitive to numerical errors and can be verified with a simple, stable calculation?

sk1p, 2020-03-09:
> Those are not small errors anymore

I didn't have time to check in the error log yet - how large were the absolute errors? Maybe we should have better assertion output here, which gives some statistics on the difference?

To reproduce, maybe try to run in a loop again?

> sensitive to numerical errors

I mean, one easy way is to just increase the dynamic range by a large margin, but this would be guaranteed to fail and also be expected. The hard part would be to find out how stable the algorithm should actually be.

I remember from the paper that they did implement Kahan summation in the minibatch part to increase stability, but I'm not sure how feasible this is.

uellue, 2020-03-09:
> how large were the absolute errors?

For example 2.4994637e-01 vs 2.4994595e-01, i.e. two orders of magnitude more than the decimal digits of precision of float32.

> The hard part would be to find out how stable the algorithm should actually be

What about "at least as good as `numpy.var()`" if we pose the task in such way that the true result is known?

In fact, LiberTEM already performs a three-level hierarchical sum if tile and partition sizes are well-chosen: First within the tile, then within the partition, and then final aggregation.

I was already thinking on how to do a complete pairwise hierarchical sum without modifying the source buffer with only `s * log(N)` memory use (s: tile size, N: number of frames) in case this becomes an issue😁.

Or just switch to `float64` 😄.

sk1p, 2020-03-09:
> What about "at least as good as numpy.var()" if we pose the task in such way that the true result is known?

Well, that would mean actually decreasing the dynamic range of our test data, as the two-pass algorithm is AFAIK the most stable of the ones compared in the paper, right?

PR #672 (2020-03-09): Perform calculation in float64
Counter stability issues with this algorithm

Closes #657

## Contributor Checklist:

* [x] I have added myself to [the creators.json file](https://github.com/LiberTEM/LiberTEM/blob/master/packaging/creators.json)
* [ ] I have added [a changelog entry](https://github.com/LiberTEM/LiberTEM/tree/master/docs/source/changelog) for my contribution
* [ ] I have added/updated test cases
commit e5a2802e (2020-03-09): Perform calculation in float64

Counter stability issues with this algorithm

Closes #657
closed 2020-03-09 by sk1p