LiberTEM/LiberTEM #660: Short installation instructions do not work

fedorov, 2020-03-01:
Re: JOSS review in https://github.com/openjournals/joss-reviews/issues/2006

Following the instructions in README, I have the following issues:
* invalid python binary (I do have python 3.7.6, and I was able to run the command using pyton3 instead)
```bash
$ virtualenv -p python3.6 ~/libertem-venv/
The path python3.6 (from --python=python3.6) does not exist
(libertem-venv)
```
* invalid pip command; I was able to run `pip install librtem`
```bash
$ pip install libertem[torch]
zsh: no matches found: libertem[torch]
(libertem-venv)
```

sk1p, 2020-03-01:
Thanks for the report! Yes, we may need to clarify that a supported Python version needs to be substituted for `python3.6` in that command.

> `zsh: no matches found: libertem[torch]`

That appears to be a quirk of zsh, should be fixable by quoting the package name argument.

fedorov, 2020-03-02:
I am sure LiberTEM users are sophisticated enough that they will figure this out without problem, so this is more for "looks" to have the front-page instructions more clean.

twentyse7en, 2020-03-04:
I would like to contribute. I got the first part, But why we need to quote the package name ?

sk1p, 2020-03-04:
> why we need to quote the package name ?

The error happens because the zsh shell used here interprets the brackets as a pattern (either glob or regular expression, I'm not sure as I use a different shell) and tries to find a file that matches the  pattern. Adding quotes removes this meaning from the brackets and passes the whole string down literally to `pip`.

sk1p, 2020-03-05:
Fixed by #664

closed 2020-03-05 by sk1p