LiberTEM/LiberTEM #671: Poster for Helmholtz Imaging Platform matchmaking workshop

uellue, 2020-03-09:
Hi all,

here's a poster draft presenting LiberTEM at the upcoming Helmholtz Imaging Platform matchmaking workshop in Heidelberg.

[LiberTEM poster for HIP-0.2-web.pdf](https://github.com/LiberTEM/LiberTEM/files/4305908/LiberTEM.poster.for.HIP-0.2-web.pdf)

Ping LiberTEM creators @sk1p @kruzaeva @woozey @AnandBaburajan @twentyse7en @jan-car @irahulcse @Iamshankhadeep @magnunor @cophus @probonopd @ozej8y @jaewshin @rafaldb

sk1p, 2020-03-09:
Note that this has just been converted into a remote video conference, so it turns out the posters won't actually be physically printed.

uellue, 2020-03-10:
> Note that this has just been converted into a remote video conference, so it turns out the posters won't actually be physically printed.

For that reason converting this issue to a landing page! :-)

uellue, 2021-02-16:
Closing since inactive :-)

closed 2021-02-16 by uellue