LiberTEM/LiberTEM #675: API changes to make integration into third-party apps easier

sk1p, 2020-03-09:
What changes would be needed to make LiberTEM more straight-forward to use from e.g. inside swift?

A possible way forward would be to use the `Context` class in swift, and do a few additions/changes:

- add `asynchronous=True` parameter to `run_udf`. This would change the return type to a new `AsyncUDFResult` (of however it will be called), from which you can access a "partial result async iterator", similar to what the `run_for_dataset` method of the `UDFRunner` returns. It could also directly have an `async cancel` method, so you don't need to invent a cancellation id anymore, if you keep the `AsyncUDFResult` instance around
- instead of registering the executor, you would register the `Context` object
- are there any other methods that need to have async variants? `Context.load` for example?
- add a `Context.detect` method; again, including async variant?

If this is done right, there should be no mention of the executor left in the plugin.

Anything I'm missing? Any more suggestions?

/cc @Brow71189

sk1p, 2020-03-09:
Something else that came up was that the return value of `detect` is a bit messy; it combines the dataset type with the parameters, so that should be untangled, too. That means the IO plugin then could create metadata that looks a bit like this:

```python
{
  "type": "HDF5",
  "parameters": {
    "ds_path": "/data/whatever",
    // ...
  },
}
```

sk1p, 2020-04-09:
The issue described in the last comment will probably be fixed by #700 /cc @AnandBaburajan

uellue, 2021-05-27:
The changes in #1011 were already very useful for Python scripting in GMS. It was quite easy to implement live plotting with the new API.

@Brow71189 how are the plans for continuing the LiberTEM integration in Swift? If there is a chance that this will continue some time this year, I'd leave this issue open for coordination. Otherwise we can perhaps close until there is specific work on this.

uellue, 2021-09-06:
Since there is no activity here and apparently nothing actionable at this time, I'll close this. We can open an Issue with specific items as needed.

closed 2021-09-06 by uellue