LiberTEM/LiberTEM #676: Include LiberTEM in "TEM" community on Zenodo.org

uellue, 2020-03-10:
Via @sk1p:

https://www.zenodo.org/communities/tem/search?page=1&size=20

We could try to include this community in the Zenodo template for LiberTEM.

sayandip18, 2020-03-11:
Does LiberTEM have any identifier other than DOI?

uellue, 2020-03-12:
@sayandip18 In what sense an identifier? For what purpose?

closed 2020-03-12 by sk1p

sayandip18, 2020-03-13:
> 
> 
> @sayandip18 In what sense an identifier? For what purpose?

Actually I was checking out how LiberTEM could have been uploaded to Zenodo and saw that there was a section titled "Related/Alternate identifiers". 

"Specify identifiers of related publications and datasets. Supported identifiers include: DOI, Handle, ARK, PURL, ISSN, ISBN, PubMed ID, PubMed Central ID, ADS Bibliographic Code, arXiv, Life Science Identifiers (LSID), EAN-13, ISTC, URNs and URLs" - this was the in the description below.

PR #677 (2020-03-12): [skip CI] Add communities to Zenodo template
Closes #676

Tested successfully on Zenodo sandbox

## Contributor Checklist:

* [x] I have added myself to [the creators.json file](https://github.com/LiberTEM/LiberTEM/blob/master/packaging/creators.json)
commit fb320de3 (2020-03-12): [skip CI] Add communities to Zenodo template

Closes #676

Tested successfully on Zenodo sandbox