LiberTEM/LiberTEM #679: Run CI with Mac OS X because of strange difference in NumPy behavior.

uellue, 2020-03-14:
When working on allowing to specify zero shifts with the blobfinder UDFs https://github.com/LiberTEM/LiberTEM-blobfinder/pull/23, I ran into a strange difference in NumPy behavior between Linux and OS X. Reporting here since it concerns our general build infrastructure and has potentially wider implications.

This function worked on Linux, but somehow modified the view or shape of the underlying array on OS X to a sub-array when views were set in `process_frame()`. It would then return only one number in the next round instead of a `(y, x)` pair, triggering failure. `zero_shift` is AUX data, `kind='nav', extra_shape=(2, )`.

```python
def get_zero_shift(self, index=None):
    if self.params.zero_shift is None:
        result = np.array((0, 0))
    elif index is None:
        # Called when masked with view
        ### This triggered the error, apparently
        result = self.params.zero_shift[:]
    else:
        # Called when not masked, in postprocess() etc.
        result = self.params.zero_shift[index]
    return result
```

This code worked:

```python
def get_zero_shift(self, index=None):
    if self.params.zero_shift is None:
        result = np.array((0, 0))
    elif index is None:
        # Called when masked with view
        ### This didn't trigger the error
        result = self.params.zero_shift
    else:
        # Called when not masked, in postprocess() etc.
        result = self.params.zero_shift[index]
    return result
```

I couldn't properly reproduce and debug it since it happened on someone else's laptop and there wasn't enough time. The unit tests passed, strangely. I'd like to work on reproducing this with a test case that runs in CI on OS X since this could potentially be a larger issue -- NumPy bug, compiler bug etc.

OS X version unknown (I'd have to ask), Python 3.7.7 from python.org in `python3 -m venv` virtual environment with fresh installation of all packages from pip.

sk1p, 2020-03-14:
Interesting. I guess the easiest way would be to add copies of the `TOXENV=py3{6,7}` jobs with `os: osx` set. Maybe even specifying a few different `osx_image` combinations (for testing different major OS X releases). Not sure how the Python is installed on travis though, if that is via the python.org package or via hombrew or similar.

uellue, 2020-04-09:
Closing here since we now test on OS X #681. I'll take care of the LiberTEM-blobfinder specific issue separately. Ref: https://github.com/LiberTEM/LiberTEM-blobfinder/pull/23

closed 2020-04-09 by uellue