LiberTEM/LiberTEM #692: Allow ds_path to be chosen from a dropdown in the HDF5 form in GUI

AnandBaburajan, 2020-03-21:
Currently when loading a HDF5 file, the 'HDF5 Dataset Path' is initially set to the largest dataset in the file. It'd be helpful if the textbox were replaced with a dropdown with multiple dataset's paths.

AnandBaburajan, 2020-03-21:
If this is a reasonable issue, I can try to work on it, but only after I'm done with my pending issues. :)

sk1p, 2020-03-21:
I'd say this is a good improvement for opening HDF5 files. Note that some files may have a very large number of items/data sets stored, see https://github.com/LiberTEM/LiberTEM/issues/440 for an example. The GUI should cope with these, too.

AnandBaburajan, 2020-03-31:
Ideally the `tileshape` textbox should be updated every time a different `ds_path` is selected, right?

> Note that some files may have a very large number of items/data sets stored, see #440 for an example. The GUI should cope with these, too.

In this case, the open file dialog fallbacks to RAW, so I think it won't be a problem.

sk1p, 2020-03-31:
> Ideally the `tileshape` textbox should be updated every time a different `ds_path` is selected, right?

I think so, although the `tileshape` tweak will soon be removed from the GUI (possibly replaced with a more appropriate tweak based on size in bytes, not shape, and negotiating the shape individually with the UDFs, #554), so it is not so important.

> In this case, the open file dialog fallbacks to RAW, so I think it won't be a problem.

Think about the edge case: how many HDF5-internal datasets can be loaded in 3 seconds? That's the case that still has to work well. If the timeout is hit, it should still be possible to open the HDF5 file "manually", by specifying the `ds_path`.

AnandBaburajan, 2020-03-31:
@sk1p So the following needs to be done:

1. Allow user to enter their own ds_path.
2. Limit number of `ds_paths` in the dropdown to `n`, in case of a huge dataset.
3. If the timeout is hit, show an error `Parameters couldn't be detected` and open the HDF5 dialog, instead of RAW.

Right?

sk1p, 2020-03-31:
I would determine `n` experimentally - like: from what point on does the GUI struggle? It is also possible to use [a searchable drop down](https://react.semantic-ui.com/modules/dropdown/), which should perform better. It could display `n` largest `ds_path`s, and then still allow to select others by searching. Does that make sense?

AnandBaburajan, 2020-03-31:
Yes, thanks!

PR #700 (2020-04-16): Update detect function and add dropdown for HDF5 ds_paths
Fixes #692 and updates datasets' `detect` function and `libertem.io.dataset.detect`. 

## Contributor Checklist:

* [x] I have added myself to [the creators.json file](https://github.com/LiberTEM/LiberTEM/blob/master/packaging/creators.json)
* [x] I have added [a changelog entry](https://github.com/LiberTEM/LiberTEM/tree/master/docs/source/changelog) for my contribution
* [x] I have added/updated documentation for all user-facing changes
* [x] I have added/updated test cases
* [x] I have included the [rebuilt production build of the cli [... cut]
closed 2020-04-16 by sk1p