LiberTEM/LiberTEM #693: Error while loading Raw dataset on windows

AyushGit123, 2020-03-26:
While loading a raw dataset produced using python script given on [website](https://libertem.github.io/LiberTEM/sample_datasets.html) , an error is displayed which says module " 'os' has no attribute 'O_DIRECT' " I do not know why this error is displayed . 
[image]

sk1p, 2020-03-26:
Which version of LiberTEM are you using? Windows does not support direct I/O, so if you are enabling it, you will get either this message with LiberTEM v0.4.1 or older, or a more friendly message with the current git version (since https://github.com/LiberTEM/LiberTEM/pull/659 by @AnandBaburajan was merged)

sk1p, 2020-03-28:
As discussed in the gitter chat, this was still LiberTEM v0.4.1, so this error is expected and replaced by a better message in #658

closed 2020-03-28 by sk1p