LiberTEM/LiberTEM #698: Button to make the server stop

twentyse7en, 2020-04-01:
An example as in Jupyter notebook :

[image]

uellue, 2020-04-09:
Is this more of a "nice to have" or does it solve a specific issue?

twentyse7en, 2020-04-09:
"nice to have"

sk1p, 2020-04-09:
> does it solve a specific issue?

Possibly the issue of stopping the LiberTEM server if you didn't start it from a terminal, for example via https://github.com/sk1p/libertem-jupyter-proxy

uellue, 2020-04-09:
> > does it solve a specific issue?
> 
> Possibly the issue of stopping the LiberTEM server if you didn't start it from a terminal, for example via https://github.com/sk1p/libertem-jupyter-proxy

Ok, good point!

twentyse7en, 2020-04-12:
> 
> Possibly the issue of stopping the LiberTEM server if you didn't start it from a terminal, for example via https://github.com/sk1p/libertem-jupyter-proxy

I would like to work on this, can you give some details regarding jupyter part.

sk1p, 2020-04-12:
You can try this yourself: run LiberTEM from jupyterlab, and kill the libertem-server process. What happens: does libertem-server stay dead, or is it revived? If it is revived, why is that? Example: the GUI keeps doing HTTP/websocket requests → this is easily fixable, other example: jupyter-proxy-server or how it's called just restarts any child processes → not so easy to fix, as it means we need to tell it to stop somehow

sk1p, 2020-04-23:
> kill libertem-server process alone

Using the process manager of the operating system, or, for example, `kill` on Linux.

twentyse7en, 2020-05-05:
Hey @AnandBaburajan can we collaborate on this issue?

AnandBaburajan, 2020-05-05:
> Hey @AnandBaburajan can we collaborate on this issue?

Sure! I'll check the details and get back to you.

AnandBaburajan, 2020-05-05:
@twentyse7en So which part do you want me to focus on? Stopping the server or the button GUI part?

twentyse7en, 2020-05-05:
No problem, I will give a try.

PR #786 (2020-06-03): Shutdown Button
Fix #698 
## Contributor Checklist:

* [x] I have added or updated my entry in [the creators.json file](https://github.com/LiberTEM/LiberTEM/blob/master/packaging/creators.json)
* [x] I have added [a changelog entry](https://github.com/LiberTEM/LiberTEM/tree/master/docs/source/changelog) for my contribution
* [ ] I have added/updated documentation for all user-facing changes
* [x] I have added/updated test cases
* [x] I have included the [rebuilt production build of the client](https://libertem.github.io/LiberTEM/contributing.html?#building-the-client) (only if
changes were mad [... cut]
closed 2020-06-03 by sk1p